RainerKuemmerle/g2o #591: “setMarginalized(false)” lead to example "ba_demo.cc" error !

chennuo0125-HIT, 2022-06-25:
I tested ba_demo.cc in g2o/examples/ba/ba_demo.cc directly, i get correct result, but when I modify “setMarginalized(true)” to “setMarginalized(false)”, i get following warning, how can i solve it ? 
[image]

RainerKuemmerle, 2022-07-02:
You also need to change the solver to accept different sizes of blocks. This seems to be a bug that solving is not rejected.

Diff would be something like
```
--- a/g2o/examples/ba/ba_demo.cpp
+++ b/g2o/examples/ba/ba_demo.cpp
@@ -113,7 +113,7 @@ int main(int argc, const char* argv[]) {
     solverName = "lm_dense6_3";
   } else {
 #ifdef G2O_HAVE_CHOLMOD
-    solverName = "lm_fix6_3_cholmod";
+    solverName = "lm_var_cholmod";
 #else
     solverName = "lm_fix6_3";
 #endif
@@ -171,7 +171,7 @@ int main(int argc, const char* argv[]) {
   for (size_t i = 0; i < true_points.size(); ++i) {
     auto v_p = std::make_shared<g2o::VertexPointXYZ>();
     v_p->setId(point_id);
-    v_p->setMarginalized(true);
+    v_p->setMarginalized(false);
     v_p->setEstimate(true_points.at(i) +

```

chennuo0125-HIT, 2022-07-04:
@RainerKuemmerle thanks for you reply :) :) :)

closed 2022-07-04 by chennuo0125-HIT