RainerKuemmerle/g2o #594: Why is the vertex pose changed when the pose is added to optimizer in sba_demo.cpp?

saikyo-kotemen-master, 2022-08-06:
I have some questions related to lines 157 through 178 of the sba_demo.cpp (a part setting camera pose as the vertex).

I am learning how to use g2o with the codes of examples in this repository.

When I extracted the vertex pose after adding it to optimizer, the vertex pose is changed.

This code adds **only the X element of translation**.
**`Vector3d trans(i * 0.04 - 1., 0, 0);`**
`pose.translation() = trans;`

But, the vertex pose after adding to optimizer includes **only the Y element of translation**.
`v_se3->setEstimate(pose);`
`optimizer.addVertex(v_se3);`
`g2o::VertexSE3Expmap* VSE3 = static_cast<g2o::VertexSE3Expmap*>(optimizer.vertex(vertex_id));`
`cout << "camera pose before optimization " << endl << static_cast<g2o::VertexSE3Expmap*>(optimizer.vertex(vertex_id))->estimate().to_homogeneous_matrix() << endl;`

The **all vertex pose** after adding to optimizer is displayed by the above cout function as follows:

> VSE3->estimate().translation() = 
 1  0  0  **0**
 0 -1  0  **1**
 0  0 -1  **0**
 0  0  0  1

Finaly, the result of optimized pose converged to [0 1 0].

However, your example code seems to work correctly because the error is reduced after optimization.

In summary,  I would have two questions about sba_demo.cpp.
- Why is the vertex pose changed when the pose is added to optimizer? 
- How does the vertex pose change? (What coordinate transformations do you use to change the vertex pose?)

RainerKuemmerle, 2022-08-20:
sba_demo is using g2o::VertexSCam. Your static_cast would change the type.
Can you please post the full code where you have doubts?

saikyo-kotemen-master, 2022-08-20:
Thank you for your reply.
My static_cast was an imitation of other sample code. As you mentioned, when I use g2o::VertexSCam instead of g2o::VertexSE3Expmap for cout function, I can confirm that the pose is outputted correctly. 
Therefore, my wrong static_cast was the cause.

`g2o::VertexSCam* VSE3 = static_cast<g2o::VertexSCam*>(optimizer.vertex(vertex_id));`
`cout << "(optimizer.vertex(vertex_id))->estimate().translation() =  " << static_cast<g2o::VertexSCam*>(optimizer.vertex(vertex_id))->estimate().translation() << endl << endl;`
`cout << "(optimizer.vertex(vertex_id))->estimate().rotation() = " << static_cast<g2o::VertexSCam*>(optimizer.vertex(vertex_id))->estimate().rotation() << endl << endl;`

Thank you very much!!

closed 2022-08-20 by RainerKuemmerle