RainerKuemmerle/g2o #595: error while loading shared libraries: libg2o_types_slam3d.so

AaboutL, 2022-08-15:
I set my CMakeLists.txt like below:
[image]
Using 
```
message("slam3d: " ${G2O_TYPES_SLAM3D})
```
I can get 
```
slam3d: /home/**/3rdparty/g2o/lib/libg2o_types_slam3d.so
```
But when execute my sample code, I get the output like below:
```
/tmp/tmp.Q6L6jGxD1s/cmake-build-debug-remotehostslam/main: error while loading shared libraries: libg2o_types_slam3d.so: cannot open shared object file: No such file or directory
```
I have tried https://github.com/RainerKuemmerle/g2o/issues/480#issuecomment-761876244（sudo ldconfig), but it does not work.
How can I solve this problem?

AaboutL, 2022-08-16:
I found a stupid solution. Since my g2o installatoin dir is not default, I write my g2o lib dir into /etc/ld.so.conf and run "ldconfig".
But I still don't know why the "core" library can work while the "types slam3d" can't. They are both found and set by cmake.

RainerKuemmerle, 2022-08-18:
You can also use the LD_LIBRARY_PATH environment variable to specify where to look for libs.
Why core library is not a problem is not so clear to me.

closed 2025-01-28 by github-actions[bot]