RainerKuemmerle/g2o #596: Configuring with `FetchContent`

TimPushkin, 2022-08-19:
I'm trying to configure a simple test project with g2o using `FetchContent` like so:
```cmake
cmake_minimum_required(VERSION 3.24)
project(testlib)

add_library(testlib SHARED library.cpp)
target_compile_features(testlib PUBLIC cxx_std_14)

include(FetchContent)
FetchContent_Declare(
        Eigen3
        GIT_REPOSITORY https://gitlab.com/libeigen/eigen.git
        GIT_TAG 3.4.0
        OVERRIDE_FIND_PACKAGE
)
FetchContent_Declare(
        g2o
        GIT_REPOSITORY https://github.com/RainerKuemmerle/g2o.git
        GIT_TAG origin/master
)
FetchContent_MakeAvailable(g2o)

target_link_libraries(testlib g2o)
```

This doesn't work: configuration finishes without errors (although there are warnings "BLA_VENDOR has been set to All but blas libraries could not be found or check of symbols failed") and I see g2o's CMake output, but during build I get `/usr/bin/ld: cannot find -lg2o: No such file or directory`. Same happens if I do `target_link_libraries(testlib g2o_core)` instead.

I also tried to use `find_package` for g2o:
```cmake
FetchContent_Declare(
        g2o
        GIT_REPOSITORY https://github.com/RainerKuemmerle/g2o.git
        GIT_TAG origin/master
        OVERRIDE_FIND_PACKAGE
)

find_package(g2o REQUIRED)

target_link_libraries(testlib g2o)
```

But this gives the same result (also with `target_link_libraries(testlib g2o_core)` instead).

Is it possible to configure g2o with `FetchProject`?

RainerKuemmerle, 2022-08-20:
I could not test with cmake 3.24 which supports the OVERRIDE_FIND_PACKAGE flag. This here works with 3.22 on Ubuntu 22.04 but has to use the internal targets as specified in the cmake files of g2o.
I believe with cmake 3.24 and a find_package call one would be able to link to g2o::core instead, i.e., the target names exported by g2o.

``` cmake
cmake_minimum_required(VERSION 3.14)
project(testlib)

add_library(testlib SHARED library.cpp)
target_compile_features(testlib PUBLIC cxx_std_14)

include(FetchContent)
FetchContent_Declare(
        Eigen3
        GIT_REPOSITORY https://gitlab.com/libeigen/eigen.git
        GIT_TAG 3.4.0
)
FetchContent_Declare(
        g2o
        GIT_REPOSITORY https://github.com/RainerKuemmerle/g2o.git
        GIT_TAG master
)
FetchContent_MakeAvailable(g2o)

target_link_libraries(testlib core)
```

RainerKuemmerle, 2022-08-20:
Tested with cmake 3.24. I think I am misunderstanding the docs here. The find_package call with OVERRIDE_FIND_PACKAGE works but the exported targets do not become visible. I guess, you have to use the internal target names with this approach.

TimPushkin, 2022-08-20:
Yeah, I also thought that the `OVERRIDE_FIND_PACKAGE` approach would work the same way as the usual install + find package approach does. I'll try the method you suggested when I get back to this in a couple of days, thanks.

TimPushkin, 2022-08-22:
> I could not test with cmake 3.24 which supports the OVERRIDE_FIND_PACKAGE flag. This here works with 3.22 on Ubuntu 22.04 but has to use the internal targets as specified in the cmake files of g2o.
> 
> ```cmake
> cmake_minimum_required(VERSION 3.14)
> project(testlib)
> 
> add_library(testlib SHARED library.cpp)
> target_compile_features(testlib PUBLIC cxx_std_14)
> 
> include(FetchContent)
> FetchContent_Declare(
>         Eigen3
>         GIT_REPOSITORY https://gitlab.com/libeigen/eigen.git
>         GIT_TAG 3.4.0
> )
> FetchContent_Declare(
>         g2o
>         GIT_REPOSITORY https://github.com/RainerKuemmerle/g2o.git
>         GIT_TAG master
> )
> FetchContent_MakeAvailable(g2o)
> 
> target_link_libraries(testlib core)
> ```

Tried it, but it doesn't work for me, and I don't think it should as g2o uses `find_package` which cannot find targets declared with `FetchContent_Declare` without `OVERRIDE_FIND_PACKAGE` as this is what this option is for. I believe, it works for you because you already have Eigen 3 installed on your PC, so g2o just finds this old installation with `find_package`.

Maybe it would be great if g2o used `FetchContent` to find Eigen if it cannot be found by `find_package` as it would not require the end users to download and build Eigen by themselves like I have to here.

TimPushkin, 2022-08-22:
I found that the following approach semi-works (it configures and builds, building g2o along the way):
```cmake
cmake_minimum_required(VERSION 3.24)
project(testlib)

add_library(testlib SHARED library.cpp)
target_compile_features(testlib PUBLIC cxx_std_14)

include(FetchContent)
FetchContent_Declare(
        Eigen3
        GIT_REPOSITORY https://gitlab.com/libeigen/eigen.git
        GIT_TAG 3.4.0
        OVERRIDE_FIND_PACKAGE
)
FetchContent_Declare(
        g2o
        GIT_REPOSITORY https://github.com/RainerKuemmerle/g2o.git
        GIT_TAG origin/master
)
FetchContent_MakeAvailable(g2o)

target_link_libraries(testlib core)
```

But I cannot figure out how should I include g2o in `library.cpp` when I use it like this. For example, I cannot use `#include <g2o/core/optimizable_graph.h>`, I have to use `#include "<my build subdir>/_deps/g2o-src/g2o/core/optimizable_graph.h"` which depends on build configuration.

RainerKuemmerle, 2022-08-22:
I think for that it would be beneficial if g2o would use target properties for setting up include directories.
At the moment, I believe we just issue on the top level cmake one include_directories call.

RainerKuemmerle, 2022-08-27:
I pushed a branch "target_include" which should help with that matter.
Probably, we can merge this to master.

CMakeLists.txt
``` cmake
cmake_minimum_required(VERSION 3.14)
project(testlib)

add_library(testlib SHARED library.cpp)
target_compile_features(testlib PUBLIC cxx_std_14)

include(FetchContent)
FetchContent_Declare(
  Eigen3
  GIT_REPOSITORY https://gitlab.com/libeigen/eigen.git
  GIT_TAG 3.4.0
)
FetchContent_Declare(
  g2o
  GIT_REPOSITORY https://github.com/RainerKuemmerle/g2o.git
  GIT_TAG target_include
)
FetchContent_MakeAvailable(g2o)

target_link_libraries(testlib core)
```
library.cpp
```c++
#include <cstdio>
#include <g2o/config.h>

namespace testlib {

void print() {
  printf("%s\n", G2O_SRC_DIR );
}

}
```
I do not see why cmake 3.24 would be required as we are not using OVERRIDE_FIND_PACKAGE.
Works for me at least also with 3.22.

TimPushkin, 2022-08-28:
For me this does not work with CMake 3.22: I'm getting the following error at configure time just as before
```
CMake Error at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Eigen3 (missing: EIGEN3_INCLUDE_DIR EIGEN3_VERSION_OK)
  (Required is at least version "3.3")
Call Stack (most recent call first):
  /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:594 (_FPHSA_FAILURE_MESSAGE)
  build/_deps/g2o-src/cmake_modules/FindEigen3.cmake:97 (find_package_handle_standard_args)
  build/_deps/g2o-src/CMakeLists.txt:486 (find_package)
```

I don't understand how this is supposed to work as we don't even call `FetchContent_MakeAvailable` for `Eigen3` so it doesn't get downloaded. But even if we do `FetchContent_MakeAvailable(Eigen3 g2o)` instead, this won't help because `Eigen3Config.cmake` will be stored in inappropriate location (`<build dir>/_deps/eigen3-build`) where `find_package` won't look for it.

I think, in your case CMake finds Eigen installed somewhere in your system.

I also tried to manually give a hint for `find_package` where to look for Eigen like so:
```cmake
FetchContent_MakeAvailable(Eigen3)
set(Eigen3_ROOT ${eigen3_BINARY_DIR})
...
FetchContent_MakeAvailable(g2o)
```

But this gives some other errors (Eigen include directory seems to be set wrong this way for some reason).

Anyway, it seems to me that g2o requires some additional `FetchContent` integration for it to work.

RainerKuemmerle, 2022-08-28:
You are right, I have Eigen3 available on the system. Since this is header only, it is easy to have it on the system.
g2o's CMakeLists should allow to specify `EIGEN3_INCLUDE_DIR` which I guess would just need to point to the folder where cmake downloaded Eigen.
I would recommend Eigen from system level due to compatibility with other libraries.

Got the following to work in a Ubuntu Jammy VM without Eigen on System level

``` cmake
cmake_minimum_required(VERSION 3.14)
project(testlib)

add_library(testlib SHARED library.cpp)
target_compile_features(testlib PUBLIC cxx_std_14)

# activate the parts you need
set(G2O_BUILD_APPS OFF CACHE BOOL "")
set(G2O_BUILD_EXAMPLES OFF CACHE BOOL "")
set(G2O_BUILD_DATA_TYPES OFF CACHE BOOL "")
set(G2O_BUILD_ICP_TYPES OFF CACHE BOOL "")
set(G2O_BUILD_SBA_TYPES OFF CACHE BOOL "")
set(G2O_BUILD_SCLAM2D_TYPES OFF CACHE BOOL "")
set(G2O_BUILD_SIM3_TYPES OFF CACHE BOOL "")
set(G2O_BUILD_SLAM2D_TYPES OFF CACHE BOOL "")
set(G2O_BUILD_SLAM2D_ADDON_TYPES OFF CACHE BOOL "")
set(G2O_BUILD_SLAM3D_TYPES OFF CACHE BOOL "")
set(G2O_BUILD_SLAM3D_ADDON_TYPES OFF CACHE BOOL "")

# activate the parts you need
set(G2O_USE_OPENGL OFF CACHE BOOL "")
set(G2O_USE_CHOLMOD OFF CACHE BOOL "")
set(G2O_USE_CSPARSE OFF CACHE BOOL "")

include(FetchContent)
FetchContent_Declare(
  Eigen3
  GIT_REPOSITORY https://gitlab.com/libeigen/eigen.git
  GIT_TAG 3.4.0
)
FetchContent_Populate(Eigen3)

set(EIGEN3_INCLUDE_DIR ${eigen3_SOURCE_DIR})
FetchContent_Declare(
  g2o
  GIT_REPOSITORY https://github.com/RainerKuemmerle/g2o.git
  GIT_TAG target_include
)
FetchContent_MakeAvailable(g2o)

target_link_libraries(testlib core)
```

This uses Eigen from the build/_deps folder, for example
```
[ 84%] Building CXX object _deps/g2o-build/g2o/solvers/pcg/CMakeFiles/solver_pcg.dir/solver_pcg.cpp.o
cd /home/ubuntu/workspace/testlib/build/_deps/g2o-build/g2o/solvers/pcg && /usr/bin/c++ -DG2O_USE_VENDORED_CERES -DUNIX -Dsolver_pcg_EXPORTS -I/home/ubuntu/workspace/testlib/build/_deps/g2o-src/g2o/solvers/pcg -I/home/ubuntu/workspace/testlib/build/_deps/g2o-src -I/home/ubuntu/workspace/testlib/build/_deps/g2o-build -isystem /home/ubuntu/workspace/testlib/build/_deps/eigen3-src -Wall -W -O3 -DNDEBUG -O3 -fPIC -msse2 -msse3 -msse4.1 -msse4.2 -std=gnu++14 -MD -MT _deps/g2o-build/g2o/solvers/pcg/CMakeFiles/solver_pcg.dir/solver_pcg.cpp.o -MF CMakeFiles/solver_pcg.dir/solver_pcg.cpp.o.d -o CMakeFiles/solver_pcg.dir/solver_pcg.cpp.o -c /home/ubuntu/workspace/testlib/build/_deps/g2o-src/g2o/solvers/pcg/solver_pcg.cpp
```

TimPushkin, 2022-08-28:
Yep, this works, thanks. I also think that it would be nice to merge `target_include` into `master`.

Although I thought g2o's CMake may be modified to do something like this internally for users' convenience:
```cmake
find_package(Eigen3 3.3 REQUIRED)
if (NOT Eigen3_FOUND)
    <use FetchContent to get Eigen>
endif ()
```

But if it won't be implemented for some reason, I can close this issue.

closed 2022-09-09 by RainerKuemmerle