RainerKuemmerle/g2o #597: Error using multiple VertexSBAPointXYZ

JunsukLee, 2022-08-24:
I want to use multiple VertexSBAPointXYZ for the Edge function (BaseBinaryEdge, BaseMultiEdge).

However, if I use multiple VertexSBAPointXYZs, a segmentation fault error occurs.

For example, using BaseBinaryEdge<2, Vector2d, VertexSBAPointXYZ, VertexSBAPointXYZ> will result in an error.

For BaseBinaryEdge<2, Vector2d, VertexSBAPointXYZ, VertexSE3Expmap> it works fine.

Also, when using BaseMultiEdge, an error occurs when using more than one VertexSBAPointXYZ.

Is VertexSBAPointXYZ only meant to be used once?

RainerKuemmerle, 2022-08-27:
Which solver do you use?
Consider using a var solver. I guess, this is a limitation of fixed size solver and the implementation of the Schur complement.

JunsukLee, 2022-08-28:
> 

The solver I used is OptimizationAlgorithmLevenberg.

RainerKuemmerle, 2022-08-28:
It would depend on the type of the BlockSolver. If you allow variable size matrices in there.
Please consider posting a minimal working example in case of follow up problems.

JunsukLee, 2022-08-28:
Thank you for answer.

I didn't know until now that blocksolver had an effect.

I am currently using the blocksolver like below.

g2o::SparseOptimizer optimizer;
g2o::BlockSolver_6_3::LinearSolverType * linearSolver; 
linearSolver = new g2o::LinearSolverEigen<g2o::BlockSolver_6_3::PoseMatrixType>();

What's the problem?

RainerKuemmerle, 2022-08-28:
You here say - rather implicitly - that you want to use the Schur complement. That one is hardcoded for binary edges where the first vertex has dimension 6 and the second one dimension 3.
Try using BlockSolverX instead of BlockSolver_6_3.

JunsukLee, 2022-08-28:
Thank you for your answer.

Thanks to your answer, I understand why the old method doesn't work.

But I want to use two customEdges at the same time like below.

(1) BaseBinaryEdge<2, Vector2d, VertexSBAPointXYZ, VertexSE3Expmap>

(2) BaseBinaryEdge<2, Vector2d, VertexSBAPointXYZ, VertexSBAPointXYZ>

If only edgeClass(2) is used, BlockSolverX seems to be able to solve it.

Is BlockSolverX a good choice if I need to use (1)+(2) at the same time?

I'd like to know if there are any performance issues with speed and accuracy.

RainerKuemmerle, 2022-08-28:
There might be a small runtime impact as the size of matrices is not known at compile time. But typically, the time needed to solve the system dominates.

JunsukLee, 2022-08-28:
thank you
It works fine.
However, the calculation time increased.

Is there a way to use Schur complement while using BlockSolverX?

RainerKuemmerle, 2022-08-29:
I forgot one point in the runtime answer. If you before did not include any edge between VertexSBAPointXYZ and another VertexSBAPointXYZ you had parts of the matrix which is block-diagonal. This is the part exploited in the Schur complement. If you add edges between two points, this part of the matrix no longer is a block diagonal. Hence, inverting it is in general a costly operation. See https://en.wikipedia.org/wiki/Schur_complement where D is the block where your points would be.
Hence, your structural change of adding constraints between points will for sure affect the runtime quite a bit because you modify the structure of your matrices.

If you have a huge number of poses you might implement Schur for them. But typically, the points are the dominant quantity.

closed 2025-01-28 by github-actions[bot]