RainerKuemmerle/g2o #600: Can't get CSparse to work with G20

tohsin, 2022-09-09:
Hey guys I have been trying to get Csparse to work with g2o and I am not sure exactly what to do
when I use this line 
```
typedef g2o::BlockSolver_6_3 BlockSolverType; // bA solver
typedef g2o::LinearSolverCSparse<BlockSolverType::PoseMatrixType> LinearSolverType;

 auto solver = new g2o::OptimizationAlgorithmLevenberg(
                g2o::make_unique<BlockSolverType>(
                        g2o::make_unique<LinearSolverType>()));
```
I get the error
```
: && /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -g -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk -dynamiclib -Wl,-headerpad_max_install_names  -o ../lib/libStereo_slam_.dylib -install_name @rpath/libStereo_slam_.dylib src/CMakeFiles/Stereo_slam_.dir/frontend.cpp.o src/CMakeFiles/Stereo_slam_.dir/camera.cpp.o src/CMakeFiles/Stereo_slam_.dir/viewer.cpp.o src/CMakeFiles/Stereo_slam_.dir/backend.cpp.o src/CMakeFiles/Stereo_slam_.dir/frame.cpp.o src/CMakeFiles/Stereo_slam_.dir/map.cpp.o  -Wl,-rpath,/usr/local/lib -Wl,-rpath,/Users/emma/dev/Pangolin/build  /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk/System/Library/Frameworks/OpenGL.framework/OpenGL.tbd  /usr/local/lib/libopencv_gapi.4.6.0.dylib  /usr/local/lib/libopencv_stitching.4.6.0.dylib  /usr/local/lib/libopencv_alphamat.4.6.0.dylib  /usr/local/lib/libopencv_aruco.4.6.0.dylib  /usr/local/lib/libopencv_barcode.4.6.0.dylib  /usr/local/lib/libopencv_bgsegm.4.6.0.dylib  /usr/local/lib/libopencv_bioinspired.4.6.0.dylib  /usr/local/lib/libopencv_ccalib.4.6.0.dylib  /usr/local/lib/libopencv_dnn_objdetect.4.6.0.dylib  /usr/local/lib/libopencv_dnn_superres.4.6.0.dylib  /usr/local/lib/libopencv_dpm.4.6.0.dylib  /usr/local/lib/libopencv_face.4.6.0.dylib  /usr/local/lib/libopencv_freetype.4.6.0.dylib  /usr/local/lib/libopencv_fuzzy.4.6.0.dylib  /usr/local/lib/libopencv_hfs.4.6.0.dylib  /usr/local/lib/libopencv_img_hash.4.6.0.dylib  /usr/local/lib/libopencv_intensity_transform.4.6.0.dylib  /usr/local/lib/libopencv_line_descriptor.4.6.0.dylib  /usr/local/lib/libopencv_mcc.4.6.0.dylib  /usr/local/lib/libopencv_quality.4.6.0.dylib  /usr/local/lib/libopencv_rapid.4.6.0.dylib  /usr/local/lib/libopencv_reg.4.6.0.dylib  /usr/local/lib/libopencv_rgbd.4.6.0.dylib  /usr/local/lib/libopencv_saliency.4.6.0.dylib  /usr/local/lib/libopencv_sfm.4.6.0.dylib  /usr/local/lib/libopencv_stereo.4.6.0.dylib  /usr/local/lib/libopencv_structured_light.4.6.0.dylib  /usr/local/lib/libopencv_superres.4.6.0.dylib  /usr/local/lib/libopencv_surface_matching.4.6.0.dylib  /usr/local/lib/libopencv_tracking.4.6.0.dylib  /usr/local/lib/libopencv_videostab.4.6.0.dylib  /usr/local/lib/libopencv_viz.4.6.0.dylib  /usr/local/lib/libopencv_wechat_qrcode.4.6.0.dylib  /usr/local/lib/libopencv_xfeatures2d.4.6.0.dylib  /usr/local/lib/libopencv_xobjdetect.4.6.0.dylib  /usr/local/lib/libopencv_xphoto.4.6.0.dylib  -framework  OpenGL  -framework  OpenGL  /Users/emma/dev/Pangolin/build/libpango_glgeometry.dylib  /Users/emma/dev/Pangolin/build/libpango_python.dylib  /Users/emma/dev/Pangolin/build/libpango_scene.dylib  /Users/emma/dev/Pangolin/build/libpango_tools.dylib  /Users/emma/dev/Pangolin/build/libpango_video.dylib  /usr/local/lib/libg2o_core.dylib  /usr/local/lib/libg2o_stuff.dylib  /usr/local/lib/libg2o_solver_csparse.dylib  /usr/local/lib/libgtest.a  /usr/local/lib/libgtest_main.a  /usr/local/lib/libglog.dylib  /usr/local/lib/libgflags.2.2.2.dylib  -lpthread  /usr/local/lib/libcxsparse.dylib  /usr/local/lib/libopencv_shape.4.6.0.dylib  /usr/local/lib/libopencv_highgui.4.6.0.dylib  /usr/local/lib/libopencv_datasets.4.6.0.dylib  /usr/local/lib/libopencv_plot.4.6.0.dylib  /usr/local/lib/libopencv_text.4.6.0.dylib  /usr/local/lib/libopencv_ml.4.6.0.dylib  /usr/local/lib/libopencv_phase_unwrapping.4.6.0.dylib  /usr/local/lib/libopencv_optflow.4.6.0.dylib  /usr/local/lib/libopencv_ximgproc.4.6.0.dylib  /usr/local/lib/libopencv_video.4.6.0.dylib  /usr/local/lib/libopencv_videoio.4.6.0.dylib  /usr/local/lib/libopencv_imgcodecs.4.6.0.dylib  /usr/local/lib/libopencv_objdetect.4.6.0.dylib  /usr/local/lib/libopencv_calib3d.4.6.0.dylib  /usr/local/lib/libopencv_dnn.4.6.0.dylib  /usr/local/lib/libopencv_features2d.4.6.0.dylib  /usr/local/lib/libopencv_flann.4.6.0.dylib  /usr/local/lib/libopencv_photo.4.6.0.dylib  /usr/local/lib/libopencv_imgproc.4.6.0.dylib  /usr/local/lib/libopencv_core.4.6.0.dylib  /Users/emma/dev/Pangolin/build/libpango_geometry.dylib  /Users/emma/dev/Pangolin/build/libtinyobj.dylib  /Users/emma/dev/Pangolin/build/libpango_plot.dylib  /Users/emma/dev/Pangolin/build/libpango_display.dylib  /Users/emma/dev/Pangolin/build/libpango_vars.dylib  /Users/emma/dev/Pangolin/build/libpango_windowing.dylib  /Users/emma/dev/Pangolin/build/libpango_opengl.dylib  /usr/local/lib/libGLEW.dylib  -framework  OpenGL  /Users/emma/dev/Pangolin/build/libpango_image.dylib  /Users/emma/dev/Pangolin/build/libpango_packetstream.dylib  /Users/emma/dev/Pangolin/build/libpango_core.dylib  /usr/local/lib/libgtest.a  /usr/local/lib/libfmt.a && :
Undefined symbols for architecture x86_64:
  "g2o::csparse_extension::cs_cholsolsymb(cs_di_sparse const*, double*, cs_di_symbolic const*, double*, int*)", referenced from:
      g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solve(g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&, double*, double*) in backend.cpp.o
  "g2o::csparse_extension::writeCs2Octave(char const*, cs_di_sparse const*, bool)", referenced from:
      g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solve(g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&, double*, double*) in backend.cpp.o
  "g2o::csparse_extension::cs_chol_workspace(cs_di_sparse const*, cs_di_symbolic const*, int*, double*)", referenced from:
      g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solveBlocks_impl(g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&, std::__1::function<void (g2o::MarginalCovarianceCholesky&)>) in backend.cpp.o
```

My cmakelist.txt
```
cmake_minimum_required(VERSION 3.22)
project(Stereo_slam_)

set(CMAKE_CXX_STANDARD 14)

#add_executable(Stereo_slam_ main.cpp src/frontend.cpp)

find_package(OpenGL REQUIRED QUIET)
include_directories(${OPENGL_INCLUDE_DIR})

list(APPEND CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake_modules)
set(EXECUTABLE_OUTPUT_PATH ${PROJECT_SOURCE_DIR}/bin)
set(LIBRARY_OUTPUT_PATH ${PROJECT_SOURCE_DIR}/lib)

find_package(fmt REQUIRED)
# gtest
find_package(GTest REQUIRED)
include_directories(${GTEST_INCLUDE_DIRS})

# glog
find_package(Glog REQUIRED)
include_directories(${GLOG_INCLUDE_DIRS})

# gflags
find_package(GFlags REQUIRED)
include_directories(${GFLAGS_INCLUDE_DIRS})

# pangolin
find_package(Pangolin REQUIRED)
include_directories(${Pangolin_INCLUDE_DIRS})

find_package(OpenCV REQUIRED)
include_directories(${OpenCV_INCLUDE_DIRS})

#Eigen
Find_Package(Eigen3 REQUIRED)
include_directories("/usr/include/eigen3")

# Sophus
find_package(Sophus REQUIRED)
include_directories(${Sophus_INCLUDE_DIRS})

# G2O
find_package(G2O REQUIRED)
include_directories(${G2O_INCLUDE_DIRS})

# csparse
find_package(CSPARSE REQUIRED)
include_directories(${CSPARSE_INCLUDE_DIR})

find_package(OpenGL REQUIRED QUIET)
include_directories(${OPENGL_INCLUDE_DIR})

set(THIRD_PARTY_LIBS
        OpenGL::GL
        Eigen3::Eigen
        ${OpenCV_LIBS}
         ${Sophus_LIBRARIES}
        ${OPENGL_LIBRARIES}
        ${Pangolin_LIBRARIES}
         ${FMT_LIBRARIES}
        ${G2O_CORE_LIBRARY} ${G2O_STUFF_LIBRARY}
        ${G20_TYPES_LIBRARY}
        ${G2O_SOLVER_CSPARSE}
        ${G20_CSPARSE_EXTENSION}
        ${G2O_SOLVER_CSPARSE_LIBRARY}
        ${GTEST_BOTH_LIBRARIES}
        ${GLOG_LIBRARIES}
        ${GFLAGS_LIBRARIES}
        pthread
        ${CSPARSE_LIBRARY}
         Sophus::Sophus

        )

enable_testing()

############### source and test ######################
include_directories(${PROJECT_SOURCE_DIR}/include)
add_subdirectory(src)
#add_subdirectory(test)
#add_subdirectory(app)
```

the file in cmake_modules to find CSparse
```
# Look for csparse; note the difference in the directory specifications!
find_path(CSPARSE_INCLUDE_DIR NAMES cs.h
  PATHS
  /usr/include/suitesparse
  /usr/include
  /opt/local/include
  /usr/local/include
  /sw/include
  /usr/include/ufsparse
  /opt/local/include/ufsparse
  /usr/local/include/ufsparse
  /sw/include/ufsparse
  PATH_SUFFIXES
  suitesparse
  )

find_library(CSPARSE_LIBRARY NAMES cxsparse libcxsparse
  PATHS
  /usr/lib
  /usr/local/lib
  /opt/local/lib
  /sw/lib
  )

include(FindPackageHandleStandardArgs)
find_package_handle_standard_args(CSPARSE DEFAULT_MSG
  CSPARSE_INCLUDE_DIR CSPARSE_LIBRARY)

```

RainerKuemmerle, 2022-09-09:
While your cmake refers to ${G20_CSPARSE_EXTENSION} this library does not show up in the linker command you pasted. Can you check whether the find script is setting this variable correctly?
Potentially other option is to use the targets file of g2o.

tohsin, 2022-09-09:
How can I use the target file of g2o sounds like it an easier fix

tohsin, 2022-09-09:
I created a folder called Thirdlibrary and cloned g2o in there, did the build and all with make  - j4.
how do I add it to my list of third party libraries I know it looks something like
```
${PROJECT_SOURCE_DIR}/Thirdparty/g2o/lib/libg2o.so
```
but I don't think I have a libg2o.so in that directory, I tried to use the CMake from Orb slam2

RainerKuemmerle, 2022-09-09:
If you anyhow want to build g2o in your project, you might want to try the fetch content approach described in https://github.com/RainerKuemmerle/g2o/issues/596#issuecomment-1229435903

Otherwise, install g2o to some folder, adapt CMAKE_INSTALL_PREFIX for that, let's say this is set to /home/user/local

On your project, a CMakeLists might look like
``` cmake
cmake_minimum_required(VERSION 3.14)
project(testlib)

find_package(g2o)

add_library(testlib SHARED library.cpp)
target_compile_features(testlib PUBLIC cxx_std_14)

target_link_libraries(testlib g2o::core)
```

Now configure this project by
`cmake -Dg2o_DIR=/home/user/local/lib/cmake/g2o ..`

tohsin, 2022-09-15:
still no luck i deleted g2o, cloned it ,did the whole 
```
mkdir build
cmake ..
make
sudo make install
```
all over again the for my project I  did
```
cmake -Dg2o_DIR=/usr/local/lib/cmake/g2o/ ..
```
 still didn't work I don't know what am doing wrong
my cmakeList.txt is 
```
cmake_minimum_required(VERSION 3.22)
project(Stereo_slam_)

set(CMAKE_CXX_STANDARD 14)

#add_executable(Stereo_slam_ main.cpp src/frontend.cpp)

find_package(OpenGL REQUIRED QUIET)
include_directories(${OPENGL_INCLUDE_DIR})

list(APPEND CMAKE_MODULE_PATH ${PROJECT_SOURCE_DIR}/cmake_modules)
set(EXECUTABLE_OUTPUT_PATH ${PROJECT_SOURCE_DIR}/bin)
set(LIBRARY_OUTPUT_PATH ${PROJECT_SOURCE_DIR}/lib)
# set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -stdlib=libstdc++ -lstdc++ -std=c++11")

find_package(fmt REQUIRED)
# gtest
find_package(GTest REQUIRED)
include_directories(${GTEST_INCLUDE_DIRS})

# glog
find_package(Glog REQUIRED)
include_directories(${GLOG_INCLUDE_DIRS})

# gflags
find_package(GFlags REQUIRED)
include_directories(${GFLAGS_INCLUDE_DIRS})

find_package(g2o)

# pangolin
find_package(Pangolin REQUIRED)
include_directories(${Pangolin_INCLUDE_DIRS})

find_package(OpenCV REQUIRED)
include_directories(${OpenCV_INCLUDE_DIRS})

#Eigen
Find_Package(Eigen3 REQUIRED)
include_directories("/usr/include/eigen3")

# Sophus
find_package(Sophus REQUIRED)
include_directories(${Sophus_INCLUDE_DIRS})

 #G2O
find_package(G2O REQUIRED)
include_directories(${G2O_INCLUDE_DIRS})

 #csparse
find_package(CSPARSE REQUIRED)
include_directories(${CSPARSE_INCLUDE_DIR})

include_directories("/usr/local/include")
include_directories(${G2O_INCLUDE_DIR})

find_package(OpenGL REQUIRED QUIET)
include_directories(${OPENGL_INCLUDE_DIR})

set(THIRD_PARTY_LIBS
        OpenGL::GL
        Eigen3::Eigen
        ${OpenCV_LIBS}
         ${Sophus_LIBRARIES}
        ${OPENGL_LIBRARIES}
        ${Pangolin_LIBRARIES}
        ${FMT_LIBRARIES}
        # g2o_core
        # g2o_stuff
        # g2o_types_sba 
        # g2o_solver_csparse
        ${G2O_CORE_LIBRARY}
        ${G2O_STUFF_LIBRARY}
        # ${G20_TYPES_LIBRARY}
        # ${G2O_SOLVER_CSPARSE}
        # ${G20_CSPARSE_EXTENSION}
        # ${G2O_SOLVER_CSPARSE_LIBRARY}
        ${GTEST_BOTH_LIBRARIES}
        ${GLOG_LIBRARIES}
        ${CSPARSE_INCLUDE_DIR}
        ${GFLAGS_LIBRARIES}
        pthread
        ${CSPARSE_LIBRARY}
         Sophus::Sophus
        )

enable_testing()

############### source and test ######################
include_directories(${PROJECT_SOURCE_DIR}/include)

add_subdirectory(src)
#add_subdirectory(test)
#add_subdirectory(app)

# ${PROJECT_SOURCE_DIR}/Thirdparty/g2o/lib
#g2o::core
```

and my find csparseis
```
# Look for csparse; note the difference in the directory specifications!
find_path(CSPARSE_INCLUDE_DIR NAMES cs.h
  PATHS
  /usr/include/suitesparse
  /usr/include
  /opt/local/include
  /usr/local/include
  /sw/include
  /usr/include/ufsparse
  /usr/local/lib/
  /opt/local/include/ufsparse
  /usr/local/include/ufsparse
  /sw/include/ufsparse
  PATH_SUFFIXES
  suitesparse
  )

find_library(CSPARSE_LIBRARY NAMES cxsparse libcxsparse
  PATHS
  /usr/lib
  /usr/local/lib
  /opt/local/lib
  /sw/lib
  )

include(FindPackageHandleStandardArgs)
find_package_handle_standard_args(CSPARSE DEFAULT_MSG
        CSPARSE_INCLUDE_DIR CSPARSE_LIBRARY)
# find_package_handle_standard_args(CSparse DEFAULT_MSG
#         CSPARSE_INCLUDE_DIR CSPARSE_LIBRARY)
```

I still keep getting the same error
```
Undefined symbols for architecture x86_64:
  "g2o::csparse_extension::cs_cholsolsymb(cs_di_sparse const*, double*, cs_di_symbolic const*, double*, int*)", referenced from:
      g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solve(g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&, double*, double*) in backend.cpp.o
  "g2o::csparse_extension::writeCs2Octave(char const*, cs_di_sparse const*, bool)", referenced from:
      g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solve(g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&, double*, double*) in backend.cpp.o
  "g2o::csparse_extension::cs_chol_workspace(cs_di_sparse const*, cs_di_symbolic const*, int*, double*)", referenced from:
      g2o::LinearSolverCSparse<Eigen::Matrix<double, 6, 6, 0, 6, 6> >::solveBlocks_impl(g2o::SparseBlockMatrix<Eigen::Matrix<double, 6, 6, 0, 6, 6> > const&, std::__1::function<void (g2o::MarginalCovarianceCholesky&)>) in backend.cpp.o
```

RainerKuemmerle, 2022-09-17:
´´´cmake
        # ${G20_TYPES_LIBRARY}
        # ${G2O_SOLVER_CSPARSE}
        # ${G20_CSPARSE_EXTENSION}
        # ${G2O_SOLVER_CSPARSE_LIBRARY}
´´´
Above all is commented anyways but please not that some (in particular the one on the csparse extension) have mixed up O (capital latter o) vs 0 (number zero).
cmake is happy to use undefined variables and replace by empty content.
Might be that fixing `${G20_CSPARSE_EXTENSION}` -> `${G2O_CSPARSE_EXTENSION}` solves your initial problem?

closed 2025-01-27 by github-actions[bot]