RainerKuemmerle/g2o #601: compile and install g2o with OpenMP, but build my program with error  undefined reference to symbol 'omp_set_lock@@OMP_3.0'

charon-cheung, 2022-09-17:
```
/usr/bin/ld: CMakeFiles/ls_slam_g2o.dir/src/main.cpp.o: undefined reference to symbol 'omp_set_lock@@OMP_3.0'
//usr/lib/x86_64-linux-gnu/libgomp.so.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
```
 `void lock() { omp_set_lock(&_lock); }` is defined in `openmp_mutex.h`, how to solve this ?

RainerKuemmerle, 2022-09-17:
https://cliutils.gitlab.io/modern-cmake/chapters/packages/OpenMP.html
Try to add openmp on your target. g2o misses to correctly export the OpenMP flags.

charon-cheung, 2022-09-19:
> https://cliutils.gitlab.io/modern-cmake/chapters/packages/OpenMP.html Try to add openmp on your target. g2o misses to correctly export the OpenMP flags.

thanks, problem solved

closed 2025-01-27 by github-actions[bot]