RainerKuemmerle/g2o #609: BaseBinaryEdge implicitly is non copyable.

Lishen1, 2022-10-24:
```BaseBinaryEdge ``` has no explicit deleted copy constructor or copy assignment operator but 
```
typename BaseFixedSizedEdge<D, E, 
VertexXi, VertexXj>::template JacobianType<
      D, VertexXi::Dimension>& _jacobianOplusXi =
      std::get<0>(this->_jacobianOplus);
```
reference made it is not possible to copy.
it invoke some problems with std containers

Lishen1, 2022-10-24:
maybe better to make something like that:
```
 T& jacobianOplusXi() {}
 const T& jacobianOplusXi() const {return ...;}

T& jacobianOplusXj() {}
 const T& jacobianOplusXj() const {return ...;}
```

RainerKuemmerle, 2022-11-01:
It was implemented like that to maintain backwards compatibility with the former member variable, to allow implementation of analytic Jacobians. For example in here
https://github.com/RainerKuemmerle/g2o/blob/master/g2o/types/slam2d/edge_se2.cpp#L63-L83

As some other internal data structures are based on pointers I wonder if the better solution is not to delete the copy c'tor and assignment operator?

Lishen1, 2022-11-01:
if it non copyable anyway, better to mark it explicitly

PR #665 (2023-03-26): Mark vertices and edges as non copyable
Fix #609
commit a3128e6a (2023-03-26): Mark vertices and edges as non copyable

Fix #609