RainerKuemmerle/g2o #612: error：buildSystem(): NaN within Jacobian for edge 0x7fffd40016c0 for vertex 1

lihuipeng1996, 2022-11-03:
When I use g2o for teb optimization, the following error occurs：**buildSystem(): NaN within Jacobian for edge 0x7fffd40016c0 for vertex 1;buildSystem(): NaN within Jacobian for edge 0x7fffd40013b0 for vertex 0.**
This error occurs during the run, not the compilation
When doing the buildgraph, I follow the build process of teb in ros,
This problem bothered me for several days, and I couldn't find a way to solve it.
I hope some friends who understand this problem can give me some ideas on how to solve it.
thanks!!!

Lishen1, 2022-11-03:
probably some numerical problem occurs in error() function due to incorrect input data or due to your implementation of error() function in graph edge.

RainerKuemmerle, 2022-11-06:
Seems similar to this issue discussed at the teb repository: https://github.com/rst-tu-dortmund/teb_local_planner/issues/384

RainerKuemmerle, 2022-11-06:
Closing, as most likely the issue is inside the implemented error function.

closed 2022-11-06 by RainerKuemmerle

lihuipeng1996, 2022-11-07:
Thank you very much. My problem has been solved. The reason is that when I was in the buildgraph, the data was wrong, which led to the failure of the buildgraph