RainerKuemmerle/g2o #616: When I did this "cmake .." ,message show ".... not found"

caixl177, 2022-12-07:
-- Looking for BLAS -- mkl.h not found
-- Looking for MKL BLAS: not found
-- Looking for Goto BLAS: not found
-- Looking for Open BLAS: not found
-- Looking for Eigen BLAS: not found
-- Looking for Eigen BLAS: not found
-- Looking for Atlas BLAS: not found
-- Looking for PhiPACK BLAS: not found
-- Looking for CXML BLAS: not found
-- Looking for DXML BLAS: not found
-- Looking for SunPerf BLAS: not found
-- Looking for SCSL BLAS: not found
-- Looking for SGIMATH BLAS: not found
-- Looking for IBM ESSL BLAS: not found
........

RainerKuemmerle, 2022-12-07:
https://github.com/RainerKuemmerle/g2o/pull/617 updates the find scripts that would silence the status messages from the initial post in this issue.

caixl177, 2022-12-08:
Thank you for your guidance！I have done this "sudo apt-get install libsuitesparse-dev",but the problem remains

caixl177, 2022-12-08:
Can I take that to mean that it works properly？

caixl177, 2022-12-08:
When I compile the program that references "g2o", it appears“/usr/local/include/g2o/core/base_fixed_sized_edge.hpp:179:14: error: ‘FixedArray {aka class ceres::internal::FixedArray<double, 3>}’ has no member named ‘fill’
   add_vertex.fill(0.);”

RainerKuemmerle, 2022-12-08:
This seems more like wrong headers get picked up to compile your program. Probably some ceres header that is installed globally?
Might be you should set G2O_USE_VENDORED_CERES to true.

caixl177, 2022-12-09:
Ok, I'll try. Thank you very much for your patience！

closed 2023-03-12 by RainerKuemmerle