RainerKuemmerle/g2o #623: /usr/lib/x86_64-linux-gnu/libEGL.so error while building on Ubunto 20.04

MF-Ahmed, 2022-12-24:
Hi I'm having the following error after build/cmake ..
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfig.cmake:27 (message):
  The imported target "Qt5::Gui" references the file

     "/usr/lib/x86_64-linux-gnu/libEGL.so"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake"

  but not all the files it references.

RainerKuemmerle, 2023-03-12:
Do you have libegl-dev installed?
But the package manager should take care to install this because of dependencies if you have qt5 dev packages installed
```
$ apt rdepends libegl-dev
libegl-dev
Reverse Depends:
  Depends: libglvnd-dev (>= 1.3.0-1)
  Depends: qtbase5-dev
  Depends: libegl1-mesa-dev
  Depends: qtbase5-gles-dev
  Depends: qtbase5-dev
  Depends: libwaffle-dev
  Depends: libgstreamer-plugins-base1.0-dev
  Depends: libegl1-mesa-dev
  Depends: libgles-dev
  Depends: libepoxy-dev
```

RainerKuemmerle, 2024-02-18:
Closing as cannot reproduce

closed 2024-02-18 by RainerKuemmerle