RainerKuemmerle/g2o #626: Python wrapper released

miquelmassot, 2023-01-16:
Hi g2o team,

I've been following your recent work in the python wrapper and I'm trying to get a Pypi release version going. I've started a wrapper repository called [g2o-python](https://github.com/miquelmassot/g2o-python) with the same spirit as opencv-python is to OpenCV.

So far, I've been able to generate a [successful release onto Pypi](https://pypi.org/project/g2o-python/), but I am interested in getting more hands on it, especially for non-Unix platforms. Are you interested in working towards this?

RainerKuemmerle, 2023-01-20:
Thank you for looking into this. I have to say that I only have a Linux machine and thus I only build the pymem branch on Linux. If you could point out issues to address for getting the python wrapper on non-Unix, I'd be happy to contribute.
I guess, one thing to start with would be to get some CI going on that branch - maybe also checking other OS?

miquelmassot, 2023-01-24:
@RainerKuemmerle I've started building wheels for manylinux, win and macos - but I have had limited time to try them. 
So far, I can say that I've also been able to build it locally from source and install it successfully, but I need to try/test the wheel distribution. 

This is the CI for uploading source distribution and wheels to Pypi: https://github.com/miquelmassot/g2o-python/blob/main/.github/workflows/wheels.yml

And this is a PR to conda-forge to add it to their repo base, which is not yet working well. https://github.com/conda-forge/staged-recipes/pull/21748

I've had to make changes to the main CMakeLists.txt in g2o to make it compatible with how pybind11 & packages work, due to currently being a source built. These lines https://github.com/miquelmassot/g2o-python/blob/main/setup.py#L19-L48 fix the issue.

miquelmassot, 2023-01-30:
I've successfully tested it on Windows and Linux with the prebuilt wheels and it's working fine. I'm going to release now version 0.0.8 with the merged PR. 

Just a note, to import the library, I need to do 
```python
from g2opy import g2opy as g2o
```
as I have not been able to change the import to a simpler one like `import g2opy as g2o`, if anyone has any solution for this I'd like to know.

I've also got a repository that I'd like to fill with examples. There's only a 2D SLAM example now.
https://github.com/miquelmassot/g2o-python-examples

RainerKuemmerle, 2023-01-30:
Is this the g2opy folder in your wrapper?
I did myself not look into packaging with pybind but it's example does not have the module with init.py, see https://github.com/pybind/cmake_example

closed 2025-01-27 by github-actions[bot]