RainerKuemmerle/g2o #636: Project webpage has out-of-date link

dkogan, 2023-02-11:
Hi. I'm looking at the web page:

  https://openslam-org.github.io/g2o.html

The "g2o on github" link points to this repo. It is correct. The "Get the Source Code!" link points to a different repo that hasn't been touched in 10 years. Presumably this is out-of-date, and should be moved to the other link.

Thanks

RainerKuemmerle, 2023-02-26:
Thanks for spotting this issue.
That page is really old. It was done when we first released g2o as a research paper.
I will make a PR towards OpenSLAM to move the link to the top as you suggested.

RainerKuemmerle, 2023-02-26:
Ah, now I understand, we have the openslam page here listed as project page.
Time for removing this. Some point in the future, we should setup a page (if needed) specific for g2o.

closed 2025-01-26 by github-actions[bot]