RainerKuemmerle/g2o #638: error with script\install-deps-windows.bat on windows

awi2d, 2023-02-26:
It seems that install-deps-windows first installs clapack and then crashes when clapack is installed.
steps to reproduce:
git clone g2o, git clone vcpkg.

```
C:\src\vcpkg>vcpkg list 

blas:x86-windows                                  1#1                 Metapackage for packages which provide BLAS
ceres:x86-windows                                 2.1.0#3             non-linear optimization package
eigen3:x86-windows                                3.4.0#2             C++ template library for linear algebra: matrice...
gflags:x86-windows                                2.2.2#5             A C++ library that implements commandline flags ...
gklib:x86-windows                                 2022-07-27#2        General helper libraries for KarypisLab.
glog:x86-windows                                  0.6.0#2             C++ implementation of the Google logging module
openblas:x64-windows                              0.3.21              OpenBLAS is an optimized BLAS library based on G...
openblas:x86-windows                              0.3.21              OpenBLAS is an optimized BLAS library based on G...
vcpkg-cmake-config:x64-windows                    2022-02-06#1
vcpkg-cmake:x64-windows                           2022-12-22
vcpkg-gfortran:x86-windows                        3#3                 Metaport to install gfortran dependencies from m...
```

clapack is not installed

```
C:\src\g2o>script\install-deps-windows.bat
The directory name is invalid.
vcpkg found at C:\src\vcpkg\vcpkg.exe...
installing eigen3 suitesparse clapack openblas ceres for triplet x86-windows...
The following packages are up-to-date:
    ceres:x86-windows
    eigen3:x86-windows
    openblas:x86-windows
error: The following packages are not installed:
    clapack:x86-windows
    suitesparse:x86-windows
internal error: D:\a\_work\1\s\src\vcpkg\commands.upgrade.cpp(193): vcpkg has crashed; no additional details are available.
Please open an issue at https://github.com/microsoft/vcpkg/issues/new?template=other-type-of-bug-report.md&labels=category:vcpkg-bug with detailed steps to reproduce the problem.
Computing installation plan...
The following packages are already installed:
    ceres[core]:x86-windows -> 2.1.0#3
    eigen3[core]:x86-windows -> 3.4.0#2
    openblas[core]:x86-windows -> 0.3.21
The following packages will be built and installed:
    clapack[core]:x86-windows -> 3.2.1#20
  * lapack[core]:x86-windows -> 2022-02-22
  * lapack-reference[blas-select,core,noblas]:x86-windows -> 3.11.0#1
  * metis[core]:x86-windows -> 2022-07-27
    suitesparse[core]:x86-windows -> 5.8.0#1
Additional packages (*) will be modified to complete this operation.
ceres:x86-windows is already installed
eigen3:x86-windows is already installed
openblas:x86-windows is already installed
Detecting compiler hash for triplet x86-windows...
Restored 1 package(s) from C:\Users\Idefix\AppData\Local\vcpkg\archives in 386 ms. Use --debug to see more details.
Installing 1/5 clapack:x86-windows...
Elapsed time to handle clapack:x86-windows: 60.08 ms
Installing 2/5 lapack-reference:x86-windows...
Building lapack-reference[blas-select,core,noblas]:x86-windows...
CMake Error at ports/lapack-reference/portfile.cmake:8 (message):
  Can't build lapack-reference if clapack is installed.  Please remove
  clapack:x86-windows, and try to install lapack-reference:x86-windows again.
Call Stack (most recent call first):
  scripts/ports.cmake:147 (include)

error: building lapack-reference:x86-windows failed with: BUILD_FAILED
error: Please ensure you're using the latest port files with `git pull` and `vcpkg update`.
Then check for known issues at:
    https://github.com/microsoft/vcpkg/issues?q=is%3Aissue+is%3Aopen+in%3Atitle+lapack-reference
You can submit a new issue at:
    https://github.com/microsoft/vcpkg/issues/new?template=report-package-build-failure.md&title=[lapack-reference]+Build+error
Include '[lapack-reference] Build error' in your bug report title, the following version information in your bug description, and attach any relevant failure logs from above.
    vcpkg-tool version: 2023-02-16-12e657924d99511514c0287ca5ce46882d3657c7
    vcpkg-scripts version: 223d33be7 2023-02-24 (2 days ago)

Please use the prefilled template from C:\src\vcpkg\installed\vcpkg\issue_body.md when reporting your issue.
```

install-deps-windoes.bat does not work as expected:
it first installs clapack:
"error: The following packages are not installed:
    clapack:x86-windows"
And then crashes because clapack is installed:
"CMake Error at ports/lapack-reference/portfile.cmake:8 (message):
Can't build lapack-reference if clapack is installed. Please remove clapack:x86-windows, and try to install lapack-reference:x86-windows again." 

And now clapack is installed for vcpkg, despite being not in the list before install-deps-windows was called:

```
C:\src\vcpkg>vcpkg list

blas:x86-windows                                  1#1                 Metapackage for packages which provide BLAS
ceres:x86-windows                                 2.1.0#3             non-linear optimization package
clapack:x86-windows                               3.2.1#20            CLAPACK (f2c'ed version of LAPACK)
eigen3:x86-windows                                3.4.0#2             C++ template library for linear algebra: matrice...
gflags:x86-windows                                2.2.2#5             A C++ library that implements commandline flags ...
gklib:x86-windows                                 2022-07-27#2        General helper libraries for KarypisLab.
glog:x86-windows                                  0.6.0#2             C++ implementation of the Google logging module
openblas:x64-windows                              0.3.21              OpenBLAS is an optimized BLAS library based on G...
openblas:x86-windows                              0.3.21              OpenBLAS is an optimized BLAS library based on G...
vcpkg-cmake-config:x64-windows                    2022-02-06#1
vcpkg-cmake:x64-windows                           2022-12-22
vcpkg-gfortran:x86-windows                        3#3                 Metaport to install gfortran dependencies from m...
```

Removing clapack with "vcpkg remove clapack:x86-windows" removes clapack from the list, but this doesnt help als install-deps-windows.bat re-installs it when used.

My system is Windows 10.

RainerKuemmerle, 2023-02-26:
The only strict dependency is eigen and you should be able to compile g2o with a bunch of solvers.
For debugging (I am not a Windows User), does it work in case you only ask to install
`eigen3 suitesparse ceres`
by changing line 18 of the script accordingly?
Thanks.

awi2d, 2023-02-27:
I had another problem with vcpkg, as installing suitesparse would either start test_zcomplexabs.exe and then stop or end because test_zcomplexabs.exe blocked some files of vcpkg.
But after changing line 18 to 'set "VcPkgLibs=eigen3 openblas ceres"' install-deps-windows.bat worked.
Thanks for the quick help.

closed 2023-02-27 by awi2d