RainerKuemmerle/g2o #642: /usr/bin/ld: /usr/lib/x86_64-linux-gnu/libQGLViewer-qt5.so: undefined reference to `qt_version_tag@Qt_5.12' collect2: error: ld returned 1 exit status

xiaofeifei-1, 2023-03-01:
/usr/bin/ld: /usr/lib/x86_64-linux-gnu/libQGLViewer-qt5.so: undefined reference to `qt_version_tag@Qt_5.12'
collect2: error: ld returned 1 exit status
make[2]: *** [g2o/examples/slam2d/CMakeFiles/slam2d_g2o.dir/build.make:151：../bin/slam2d_g2o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:3103：g2o/examples/slam2d/CMakeFiles/slam2d_g2o.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务...

version: [20230223_git](https://github.com/RainerKuemmerle/g2o/releases/tag/20230223_git)
platform:ubuntu20.04

RainerKuemmerle, 2023-03-03:
I cannot reproduce this in a VM running Ubuntu 20.04.5 LTS. Building passes fine.

Do you have all packages on your system installed from the Ubuntu Apt repositories. Is Qt5 installed via apt?
See for example this older issue where conda interfered with system libraries:
https://github.com/RainerKuemmerle/g2o/issues/446#issuecomment-708596131

Please provide more output. Full make log and CMakeCache.

```
ubuntu@focal:~/workspace/g2o/build$ ldd /usr/lib/x86_64-linux-gnu/libQGLViewer-qt5.so
	linux-vdso.so.1 (0x00007fff19b16000)
	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007f4fb0360000)
	libGLU.so.1 => /lib/x86_64-linux-gnu/libGLU.so.1 (0x00007f4fb02ee000)
	libQt5OpenGL.so.5 => /lib/x86_64-linux-gnu/libQt5OpenGL.so.5 (0x00007f4fb0292000)
	libQt5Widgets.so.5 => /lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f4fafbf7000)
	libQt5Gui.so.5 => /lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f4faf60c000)
	libQt5Xml.so.5 => /lib/x86_64-linux-gnu/libQt5Xml.so.5 (0x00007f4faf5ca000)
	libQt5Core.so.5 => /lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f4faf07f000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f4faee9d000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f4faed4e000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f4faed33000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f4faeb41000)
	libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007f4faea89000)
	libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007f4faea53000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f4faea30000)
	libpng16.so.16 => /lib/x86_64-linux-gnu/libpng16.so.16 (0x00007f4fae9f8000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f4fae9dc000)
	libharfbuzz.so.0 => /lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f4fae8d8000)
	libicui18n.so.66 => /lib/x86_64-linux-gnu/libicui18n.so.66 (0x00007f4fae5d7000)
	libicuuc.so.66 => /lib/x86_64-linux-gnu/libicuuc.so.66 (0x00007f4fae3f1000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f4fae3eb000)
	libpcre2-16.so.0 => /lib/x86_64-linux-gnu/libpcre2-16.so.0 (0x00007f4fae367000)
	libdouble-conversion.so.3 => /lib/x86_64-linux-gnu/libdouble-conversion.so.3 (0x00007f4fae351000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f4fae228000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f4fb049c000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007f4fae0e9000)
	libfreetype.so.6 => /lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f4fae02a000)
	libgraphite2.so.3 => /lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f4fadffd000)
	libicudata.so.66 => /lib/x86_64-linux-gnu/libicudata.so.66 (0x00007f4fac53c000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f4fac4c9000)
	libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f4fac49d000)
	libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007f4fac497000)
	libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f4fac48f000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f4fac475000)
ubuntu@focal:~/workspace/g2o/build$ nm -D -C /lib/x86_64-linux-gnu/libQt5Core.so.5 | grep version_tag
00000000003663b2 R qt_version_tag
00000000003663b7 R qt_version_tag
00000000003663b1 R qt_version_tag
00000000003663b5 R qt_version_tag
00000000003663bb R qt_version_tag
00000000003663b4 R qt_version_tag
00000000003663b9 R qt_version_tag
00000000003663b8 R qt_version_tag
00000000003663b6 R qt_version_tag
00000000003663b3 R qt_version_tag
00000000003663bc R qt_version_tag
00000000003663b0 R qt_version_tag
00000000003663ba R qt_version_tag
00000000003663bc R qt_version_tag_5_0
00000000003663bb R qt_version_tag_5_1
00000000003663b2 R qt_version_tag_5_10
00000000003663b1 R qt_version_tag_5_11
00000000003663b0 R qt_version_tag_5_12
00000000003663ba R qt_version_tag_5_2
00000000003663b9 R qt_version_tag_5_3
00000000003663b8 R qt_version_tag_5_4
00000000003663b7 R qt_version_tag_5_5
00000000003663b6 R qt_version_tag_5_6
00000000003663b5 R qt_version_tag_5_7
00000000003663b4 R qt_version_tag_5_8
00000000003663b3 R qt_version_tag_5_9
```

piedaochuan1234, 2023-03-06:
@xiaofeifei-1 sovled by [](https://github.com/Shubodh/MR-project1-pgo/issues/5#issuecomment-708898269)

RainerKuemmerle, 2023-03-12:
Probably conda installation leads to conflicts.

closed 2023-03-12 by RainerKuemmerle