RainerKuemmerle/g2o #643: Cholesky failure

Lishen1, 2023-03-02:
Cholesky failure, writing debug.txt (Hessian loadable by  Octave) message. while debug.txt can be solved by numpy or even such workaround works:
```
 Eigen::SimplicialLLT<SparseMatrix> solver;
      solver.analyzePattern(_sparseMatrix);
      solver.factorize(_sparseMatrix);

      if (this->writeDebug()) {
        std::cerr << "Cholesky failure, writing debug.txt (Hessian loadable by "
                     "Octave)"
                  << std::endl;
        A.writeOctave("debug.txt");
      }

      if (solver.info() == Eigen::Success) {
        return true;
      }
```
[debug.txt](https://github.com/RainerKuemmerle/g2o/files/10870439/debug.txt)

RainerKuemmerle, 2023-03-02:
```
octave:1> load("debug.txt")
octave:2> L = chol(debug)
error: chol: input matrix must be positive definite
```
Using the above debug.txt file.
What do you mean by it can be solved with numpy?
Your matrix does also not appear to be symmetric. Make sure that your information matrices are symmetric and positive semi definite.

Your code modification is meant for https://github.com/RainerKuemmerle/g2o/blob/master/g2o/solvers/eigen/linear_solver_eigen.h#L124:L134
That would just deactivate print the failure and writing the debug independent of the solvers success. I do not entirely get the idea behind that?

Lishen1, 2023-03-02:
> That would just deactivate print the failure and writing the debug independent of the solvers success. I do not entirely get the idea behind that?

i mean if we are already in failure branch //if (_cholesky.info() != Eigen::Success))
the same _sparseMatrix where _cholesky failed can be solved with 
```solver.analyzePattern(_sparseMatrix);  solver.factorize(_sparseMatrix);```
```if (solver.info() == Eigen::Success) {    return true;   }```
will be successful.

> What do you mean by it can be solved with numpy?

i load debug.txt into dense numpy array and it can be decompose too.
> Your matrix does also not appear to be symmetric. Make sure that your information matrices are symmetric and positive semi definite.

probably yes, when covariance matrix become too big some numerical instabilities appear

RainerKuemmerle, 2023-03-02:
Reading the numpy doc at https://numpy.org/doc/stable/reference/generated/numpy.linalg.cholesky.html

> In addition, only the lower-triangular and diagonal elements of a are used.

So I assume by this you make the matrix implicitly symmetric and that one is also positive definite. We can do the same in octave:
```
octave:1> load "debug.txt"
octave:2> lower = tril(debug, -1);
octave:3> reconstructed = lower + diag(diag(debug)) + transpose(lower);
octave:4> chol(reconstructed);
```

g2o uses the upper instead of the lower to prepare the input for cholesky, see
https://github.com/RainerKuemmerle/g2o/blob/master/g2o/solvers/eigen/linear_solver_eigen.h#L56:L57
https://eigen.tuxfamily.org/dox/classEigen_1_1LLT.html

I tried to reproduce this in octave as well:
```
octave:1> load "debug.txt"
octave:2> upper = triu(debug, 1);
octave:3> reconstructed_upper = upper + diag(diag(debug)) + transpose(upper);
octave:4> chol(reconstructed_upper);
error: chol: input matrix must be positive definite
```
This matrix has an eigen value which is negative: -7.8051e+03, for the reconstruction with the lower part all eigen values are positive.

About calling it again:
You use Eigen's default to only use the lower triangle. But the matrix is filled by g2o only on the upper triangle. So I guess, the cholesky solver only operates on the diagonal? But I did not check this in detail.

Can you check whether there is in issue in your problem constructions?

RainerKuemmerle, 2023-03-03:
Closing this issue as the problem was in the provided information matrices.

closed 2023-03-03 by RainerKuemmerle