RainerKuemmerle/g2o #646: g2o uses Eigen as if it is compiled with C++14 even if a newer standard is available

TimPushkin, 2023-03-04:
I haven't figured out exactly how this works, but [this line in g2o's main CMake file](https://github.com/RainerKuemmerle/g2o/blob/bd829227e2075287ec0f61ea1682ef1ba1d3e7cb/CMakeLists.txt#L481) makes g2o use Eigen's headers as if they will be compiled with only C++14 supported, even if C++17 or newer is available.

The exact problem I've stumbled upon is that g2o uses `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` macros which generates destructors that Address Sanitizer reports as causing buffer overflow. The thing is, my project uses C++17 which is configured as a target property of my library, and with C++17 available `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` [should do nothing](https://gitlab.com/libeigen/eigen/-/blob/master/Eigen/src/Core/util/Memory.h#L845). However, because of the CMake line mentioned above, in g2o this macros generates faulty code (I'm not sure if it is truly faulty or if this is just a false positive, but I would like to get rid of this anyway).

I see several approaches of resolving this:
1. If g2o is built as a top-level project, set `CMAKE_CXX_STANDARD` as a cached variable so it can be overridden, or if g2o is not a top-level project check if `CMAKE_CXX_STANDARD` is set. And then check if it is set to a value >= 14. I'm not sure what to do when g2o is not a top-level project and `CMAKE_CXX_STANDARD` is not set though -- and this is exactly my case as I control the version of the C++ standard with the corresponding target property of my library. Though, of course, I can set `CMAKE_CXX_STANDARD` just when I set up g2o.
2. Set C++ standard as a target property -- [it is a recommended way of doing this in the modern CMake](https://cmake.org/cmake/help/latest/manual/cmake-compile-features.7.html#id6). This will be more verbose as g2o has a lot of separate targets.

I haven't tested these approaches, but I hope they will work as expected: set the minimal required C++ standard version for g2o to C++14 and let Eigen use newer versions if they are available.

TimPushkin, 2023-03-05:
I've just noticed that g2o's core target actually has a C++14 requirement [here](https://github.com/RainerKuemmerle/g2o/blob/bd829227e2075287ec0f61ea1682ef1ba1d3e7cb/g2o/core/CMakeLists.txt#L58). This should be enough by itself, so I think the lines that set `CMAKE_CXX_STANDARD` and `CMAKE_CXX_STANDARD_REQUIRED` can just be removed.

RainerKuemmerle, 2023-03-05:
I was planing on moving to c++17 on master branch.
But I am a little confused now about EIGEN_MAKE_ALIGNED_OPERATOR_NEW.

Looking when EIGEN_HAS_CXX17_OVERALIGN would be 1. It's somewhat a bit more narrow than just c++17?
https://gitlab.com/libeigen/eigen/-/blob/master/Eigen/src/Core/util/Macros.h#L752

The doc on https://eigen.tuxfamily.org/dox/group__TopicStructHavingEigenMembers.html seems to simplify this?

Also, when a file is compiled with c++14, EIGEN_MAKE_ALIGNED_OPERATOR_NEW would expand to actual code but when you include with c++17 it would be empty?

TimPushkin, 2023-03-05:
`EIGEN_MAKE_ALIGNED_OPERATOR_NEW` should expand to nothing when the project is compiled in C++17 mode and the compiler supports ["Dynamic memory allocation for over-aligned data" feature of C++17](https://www.open-std.org/jtc1/sc22/wg21/docs/papers/2016/p0035r4.html) and to the actual code otherwise. On my system `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` expands to the mentioned faulty code when compiled for C++14 and to nothing when compiled for C++17.

With `EIGEN_HAS_CXX17_OVERALIGN` Eigen seems to check just that. To be 100% safe when moving to C++17 it is best to either leave `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` or place `static_assert(EIGEN_HAS_CXX17_OVERALIGN)` somewhere (`config.h`?). But CMake's developers seem to think that it is save to assume that if a compiler supports C++17, it supports all the required features, so they don't provide ways to request individual features like they do for C++ 11 and 14.

RainerKuemmerle, 2023-03-07:
Thanks for the further explanation.
This means the branch switching to c++17 would address the issue and we leave the `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` in place for now.

RainerKuemmerle, 2023-03-19:
I merged to master the change to use c++17 and require Eigen 3.4 as well. By this I close this issue, let me know if further changes would be needed. Thanks.

closed 2023-03-19 by RainerKuemmerle