RainerKuemmerle/g2o #647: while compiling with cmake on ubuntu,  g2o_viewer failed after the 'make' command

ligang305, 2023-03-07:
[image]

RainerKuemmerle, 2023-03-07:
What's your cmake cache. Make sure the library for qglviewer is found and still in your cmake.
Built by calling make VEROSE=1 to see the linker command line.

ligang305, 2023-03-09:
Thx, I have changed to the following branch and there is no error.
[image]

RainerKuemmerle, 2023-03-12:
I am a bit puzzled how that solved the problem.
Please re-open in case you want to follow up to understand the root-cause of the problem. Thanks.

closed 2023-03-12 by RainerKuemmerle