RainerKuemmerle/g2o #648: g2o should build its documentation instead of including the binary g2o.pdf in the repo

dkogan, 2023-03-10:
Hi. I'm working on the Debian package. There will be a series of issues. Let me know if email, or something, is more appropriate.

The patches in the debianization live here: https://salsa.debian.org/science-team/g2o/-/tree/master/debian/patches

The first patch is to build g2o.pdf instead of including the non-editable binaries (the g2o.pdf and the figures in doc/pics) in the repo. Patch: https://salsa.debian.org/science-team/g2o/-/blob/master/debian/patches/documentation-built-with-pdflatex.patch

RainerKuemmerle, 2023-03-10:
Thank you. I created a branch with a commit reflecting your patch file.
Is it ok to use your name+email as author or should I change that for the commit reflecting your patch?

dkogan, 2023-03-11:
You can import the commit exactly as is (including my name+email), or
you can modify it as you like. This is small, and I don't care a whole
lot.

Thanks

commit 7b2c745c (2023-03-12): Compile pdf documentation

Fix #648
closed 2023-03-12 by RainerKuemmerle