RainerKuemmerle/g2o #649: g2o shared libraries need to have an SONAME, SOVERSION

dkogan, 2023-03-10:
Patch is here:

https://salsa.debian.org/science-team/g2o/-/blob/master/debian/patches/libraries-have-versioned-soname.patch

This defines the ABI version. This is encoded in the shared object and the package names. Dependencies (other shared objects, executables and packages) use this to ensure that they're compatible. Hopefully you don't break the ABI with each release. If the ABI changes, the SOVERSION will need a bump

RainerKuemmerle, 2023-03-12:
```
set(G2O_LIB_VERSION   "0.1.0" CACHE STRING "g2o library version")
set(G2O_LIB_SOVERSION "0.1"   CACHE STRING "g2o library soversion")
```
Does it make sense to use semantic version as above? I thought, this would allow bugfix releases keeping the SO name stable - if I understood correctly.
Furthermore, I could use a higher major version number on the pymem branch indicating API/ABI incompatibility.

closed 2023-03-18 by RainerKuemmerle

dkogan, 2023-03-19:
Thank you very much for adding these. The versions you mention promise a stable ABI for 0.1.x. If the ABI has changed, you bump to 0.2.x and so on. I think that's reasonable. Thanks!

PR #654 (2023-03-18): Add SO_VERSION and VERSION
Using semantic version numbers

Fix #649
commit f902c00f (2023-03-18): Add SO_VERSION and VERSION (#654)

Using semantic version numbers

Fix #649