RainerKuemmerle/g2o #650: Which binaries should be distributed into the Debian package?

dkogan, 2023-03-10:
Hi. I currently see these in `/usr/bin`:

- g2o_anonymize_observations_rd
- g2o_rd
- g2o_simulator2d_rd
- g2o_simulator3d_rd

Running any of them doesn't do anything obviously useful, but I don't know what they're supposed to do (I'm packaging this for somebody else, and don't use g2o myself). Should these be shipped?

In `/usr/lib` are lots of files named `libg2o_XXXXX_rd.so`, and I ship everything. Questions:

- What is the `_rd` suffix on everything? Is it supposed to be there?
- Which libraries would the end user link against? Would they pick and choose between all the various `libg2o_XXXXX_rd.so`? Some of them have `NEEDED` tags to depend on other libraries, and some do not. I would just like to confirm that distribution all of the libraries in `/usr/lib` is the right thing to do.
 
Thanks!

RainerKuemmerle, 2023-03-10:
### Binaries
About the binaries. The g2o binary allows to run the optimizer on text files representing an optimization problem. There is also a viewer depending on libqglviewer and qt (not sure about their versions and situation in Debian) which allows to view the problems graphically as well.
g2o_simulator2d and g2o_simulator3d allow to generate such optimization problem files by running a fairly simple optimizer. This allows to test the framework on example data.
Could be useful to include all those binaries.

### Libraries
g2o tries to be sort of modular. Hence, the set of libraries with dependencies between them.
User programming against the library may only need a subset of the libraries to achieve their needs. Some might reuse the types (what kind of optimization problem to solve), extend, or write their own types. Solvers are typically used from one library. This depends on what dependencies are available at build time.
The most efficient solvers are provided by libsuitesparse-dev.

I guess you do not include examples (both libs and bins)?

### About the _rd suffix
This is due to https://github.com/RainerKuemmerle/g2o/blob/master/CMakeLists.txt#L31
We added this once because we when building put the libs and binary into lib and bin folder on top-level.
Probably it would be better if we remove this and place everything in the build folder instead of using one folder with suffixes to indicate the build type.
Short term workaround for you would be to patch that suffix to be an empty string?

dkogan, 2023-03-11:
Hi. Thanks for the detailed notes.

I will ship all the executables. Thanks for confirming that they're all useful.

I will also include all the non-solver libraries. It just wasn't clear
to me how they were meant to be linked in.

> Solvers are typically used from one library. This depends on what
> dependencies are available at build time. The most efficient solvers
> are provided by libsuitesparse-dev.

Everything is available in Debian. Should I only include the suitesparse
solvers (cholmod, csparse)? Probably I should include all of them;
right?

> I guess you do not include examples (both libs and bins)?

I will include their sources as documentation, but not the binaries.
Depending on what your build system does, I might still build them when
the package is built. Is this reasonable? Are the example binaries
useful to have with each install?

> About the _rd suffix
>
> This is due to
> https://github.com/RainerKuemmerle/g2o/blob/master/CMakeLists.txt#L31
> We added this once because we when building put the libs and binary
> into lib and bin folder [...]

OK, can we not do this anymore? Once stuff is installed, nothing should
have the _rd suffix. Currently Debian is in a freeze while the new
release is being cleaned up, so new packages such as this cannot be
uploaded anyway. If you could fix this in the next few months, that
would be really great.

Thanks!

RainerKuemmerle, 2023-03-12:
> I will ship all the executables. Thanks for confirming that they're all useful. I will also include all the non-solver libraries. It just wasn't clear to me how they were meant to be linked in.

g2o_hierarchical might not be useful in a lot of cases. I might move it to the examples folder instead.

> Everything is available in Debian. Should I only include the suitesparse solvers (cholmod, csparse)? Probably I should include all of them; right?

It makes sense to include cholmod for sure. CSparse and Eigen solver are very similar because the Eigen code is based on the CSparse code afaik. For now, I'd recommend to include all. In the future, I might drop CSparse solver.

> I will include their sources as documentation, but not the binaries. Depending on what your build system does, I might still build them when the package is built. Is this reasonable? Are the example binaries useful to have with each install?

Considering examples as documentation makes sense. There is a CMake option called G2O_BUILD_EXAMPLES to not build the examples at all.
https://github.com/RainerKuemmerle/g2o/blob/master/CMakeLists.txt#L285

> OK, can we not do this anymore? Once stuff is installed, nothing should have the _rd suffix. Currently Debian is in a freeze while the new release is being cleaned up, so new packages such as this [...]

I will do that.

RainerKuemmerle, 2023-03-12:
https://github.com/RainerKuemmerle/g2o/pull/653

The above PR should fix the _rd suffix and also less binaries in case we build with G2O_BUILD_EXAMPLES=OFF

RainerKuemmerle, 2023-03-12:
By configuring with `cmake -DG2O_BUILD_EXAMPLES=OFF ..` we obtain on the linked branch

```
$ ls -1 bin
g2o
g2o_simulator2d
g2o_simulator3d
g2o_viewer
```
Those are also the binaries which are most useful as they allow to simulate, view, and optimize 2D and 3D SLAM problems.

dkogan, 2023-03-19:
Thanks much. I'm going to come back to this in a bit, and will comment if anything else is unclear.

dkogan, 2023-08-05:
Hi. I'm looking at this again now. It looks like the last release is from 2023/02, and lots of stuff has happened since then; including the merge you mentioned above: #653. Can you make a new release? Or should I package the latest tag with these patches cherry-picked?

Thanks

RainerKuemmerle, 2023-08-06:
I released a new version with the state of the master branch today: https://github.com/RainerKuemmerle/g2o/releases/tag/20230806_git

dkogan, 2023-08-07:
Thanks for doing that. I just uploaded the packages. Should pass the NEW queue, and will end up in the main Debian archive in a few weeks probably.

closed 2023-08-07 by dkogan

RainerKuemmerle, 2023-08-07:
@dkogan Awesome. Looking forward to that!