RainerKuemmerle/g2o #655: g2o_viewer can't show id

w8128835, 2023-03-16:
Hi,g2o_viewer is a really nice tool to visualize vertex and edges,but I met a little problem when i use g2o_viewer.
I want to check the impact of specific vertexs and edges that I set on the optimization process ,so I set enable show id in g2o_viewer.But  it doesn't work. Can you help me?
 I can't see id in it ,like the picture shows below:
[image]

RainerKuemmerle, 2023-03-18:
Thanks for reporting!
This seems indeed very broken. I'm not sure when we broke that or if it was once working at all.

As a workaround for vertex IDs you can add a user data with VERTEX_TAG as payload.
see `setUserData` on your vertex. You would have to pass a VertexTag with appropriate ID.
In case of files, here an example:

```
VERTEX_SE2 1100 0 0 0
VERTEX_TAG 1100
VERTEX_SE2 1101 0 0 1.5708
VERTEX_TAG 1101
VERTEX_SE2 1102 6.12323e-17 1 1.5708
VERTEX_TAG 1102
VERTEX_SE2 1103 1.22465e-16 2 1.5708
VERTEX_TAG 1103
VERTEX_SE2 1104 1.83697e-16 3 1.5708
VERTEX_TAG 1104
VERTEX_SE2 1105 1.83697e-16 3 0
VERTEX_TAG 1105
VERTEX_SE2 1106 1 3 0
VERTEX_TAG 1106
VERTEX_SE2 1107 1 3 -1.5708
VERTEX_TAG 1107
VERTEX_SE2 1108 1 3 0
VERTEX_TAG 1108
VERTEX_SE2 1109 1 3 -1.5708
VERTEX_TAG 1109
VERTEX_SE2 1110 1 2 -1.5708
VERTEX_TAG 1110
EDGE_SE2 1100 1101 0.00654639 0.00601635 1.57731 500 0 0 500 0 5000
EDGE_SE2 1101 1102 0.916309 0.00732141 -0.00303 500 0 0 500 0 5000
EDGE_SE2 1102 1103 1.01335 -0.0370268 0.000144468 500 0 0 500 0 5000
EDGE_SE2 1103 1104 1.0472 -0.0244555 0.0166109 500 0 0 500 0 5000
EDGE_SE2 1104 1105 -0.0469325 -0.0295466 -1.54978 500 0 0 500 0 5000
EDGE_SE2 1105 1106 0.972037 -0.114447 -0.011725 500 0 0 500 0 5000
EDGE_SE2 1106 1107 -0.0397442 0.0241397 -1.57969 500 0 0 500 0 5000
EDGE_SE2 1107 1108 -0.0151868 0.0455811 1.56397 500 0 0 500 0 5000
EDGE_SE2 1108 1109 -0.00542498 -0.094311 -1.57486 500 0 0 500 0 5000
EDGE_SE2 1110 1100 1.99398 -1.01659 1.56873 500 0 0 500 0 5000
EDGE_SE2 1109 1110 0.952624 -0.102924 0.0205771 500 0 0 500 0 5000
```

Result
[image]

To fix this issue, we would have to add routines to draw the ID in the draw functions of each class.
I wonder if the TAG approach is not more general for vertex IDs. That would mean, I remove this generic broken option and the user has to add VertexTag in case IDs shall be visualized.

w8128835, 2023-03-20:
@RainerKuemmerle Sorry for lately reply.I tried the way you taught me this morning,it is really useful.I think it can help people to see what happend during optimization,it is very helpful.Thank you very much!

closed 2023-03-20 by w8128835