RainerKuemmerle/g2o #659: Build Error Glog

goktugyildirim4d, 2023-03-23:
When I try to use g2o::EdgeSE3ProjectXYZOnlyPose with the version of the last release, I face the build error. Do you have any idea?

`
/usr/bin/ld: CMakeFiles/g2o_optimizers.dir/src/only_pose_ba_solver.cpp.o: in function `ceres::internal::FixedArray<double, 6ul, std::allocator<double> >::operator[](unsigned long)':
only_pose_ba_solver.cpp:(.text._ZN5ceres8internal10FixedArrayIdLm6ESaIdEEixEm[_ZN5ceres8internal10FixedArrayIdLm6ESaIdEEixEm]+0xad): undefined reference to `google::LogMessageFatal::LogMessageFatal(char const*, int, google::CheckOpString const&)'
/usr/bin/ld: only_pose_ba_solver.cpp:(.text._ZN5ceres8internal10FixedArrayIdLm6ESaIdEEixEm[_ZN5ceres8internal10FixedArrayIdLm6ESaIdEEixEm]+0xb9): undefined reference to `google::LogMessage::stream()'
/usr/bin/ld: only_pose_ba_solver.cpp:(.text._ZN5ceres8internal10FixedArrayIdLm6ESaIdEEixEm[_ZN5ceres8internal10FixedArrayIdLm6ESaIdEEixEm]+0xc5): undefined reference to `google::LogMessageFatal::~LogMessageFatal()'
/usr/bin/ld: only_pose_ba_solver.cpp:(.text._ZN5ceres8internal10FixedArrayIdLm6ESaIdEEixEm[_ZN5ceres8internal10FixedArrayIdLm6ESaIdEEixEm]+0x106): undefined reference to `google::LogMessageFatal::~LogMessageFatal()'
/usr/bin/ld: CMakeFiles/g2o_optimizers.dir/src/only_pose_ba_solver.cpp.o: in function `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >* google::MakeCheckOpString<unsigned long, unsigned long>(unsigned long const&, unsigned long const&, char const*)':
only_pose_ba_solver.cpp:(.text._ZN6google17MakeCheckOpStringImmEEPNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_RKT0_PKc[_ZN6google17MakeCheckOpStringImmEEPNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_RKT0_PKc]+0x37): undefined reference to `google::base::CheckOpMessageBuilder::CheckOpMessageBuilder(char const*)'
/usr/bin/ld: only_pose_ba_solver.cpp:(.text._ZN6google17MakeCheckOpStringImmEEPNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_RKT0_PKc[_ZN6google17MakeCheckOpStringImmEEPNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_RKT0_PKc]+0x61): undefined reference to `google::base::CheckOpMessageBuilder::ForVar2()'
/usr/bin/ld: only_pose_ba_solver.cpp:(.text._ZN6google17MakeCheckOpStringImmEEPNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_RKT0_PKc[_ZN6google17MakeCheckOpStringImmEEPNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_RKT0_PKc]+0x7f): undefined reference to `google::base::CheckOpMessageBuilder::NewString[abi:cxx11]()'
/usr/bin/ld: only_pose_ba_solver.cpp:(.text._ZN6google17MakeCheckOpStringImmEEPNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_RKT0_PKc[_ZN6google17MakeCheckOpStringImmEEPNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_RKT0_PKc]+0x8f): undefined reference to `google::base::CheckOpMessageBuilder::~CheckOpMessageBuilder()'
/usr/bin/ld: only_pose_ba_solver.cpp:(.text._ZN6google17MakeCheckOpStringImmEEPNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_RKT0_PKc[_ZN6google17MakeCheckOpStringImmEEPNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEERKT_RKT0_PKc]+0xb6): undefined reference to `google::base::CheckOpMessageBuilder::~CheckOpMessageBuilder()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/g2o_optimizers.dir/build.make:277: g2o_optimizers] Error 1
make[1]: *** [CMakeFiles/Makefile2:83: CMakeFiles/g2o_optimizers.dir/all] Error 2
make: *** [Makefile:91: all] Error 2

`

RainerKuemmerle, 2023-03-25:
I believe this is due to your build picking up ceres headers from the system instead of from g2o. I guess, it was some sort of bad practice on my end to include ceres headers without changing namespace to avoid linker issues.
Master branch hopefully clears this issue. Can you please check if this works for you.

goktugyildirim4d, 2023-03-27:
After I purged Ceres and re-built g2o, It solved. Thanks.

closed 2023-03-27 by goktugyildirim4d