RainerKuemmerle/g2o #667: Retrieve posterior uncertainty after optimisation?

miquelmassot, 2023-03-29:
Hi!

I'd like to know if there's a way to retrieve the posterior uncertainty (or information matrix) after optimisation has been done. I can't seem to find information about this, although some old documentation seemed to state the method `.covariance()` on BaseVertex class, which no longer exists. Does anyone have information about this?

RainerKuemmerle, 2023-03-29:
After optimization has been run, one can retrieve the estimate of the uncertainty (as approximated by the Hessian) via computeMarginals on the SparseOptimizer class.
I admit that documentation is outdated. In the past, the covariance was stored on each vertex which caused a high memory footprint.
Now instead a sparse matrix has to be passed for storing this information on demand. You have to consider that the covariance is expressed in the manifold space.

The class below shows research code for computing a condensed graph given a set of nodes/edges
[image]
https://github.com/RainerKuemmerle/g2o/blob/master/g2o/examples/g2o_hierarchical/edge_labeler.cpp 
Paper: http://ais.informatik.uni-freiburg.de/publications/papers/grisetti10icra.pdf

Furthermore, note that computing the inverse takes significantly more time as just solving the system.

tudoroancea, 2023-04-21:
Hi @miquelmassot ! Did you end up finding a solution to you problem with the python wrapper? I am interested as well

RainerKuemmerle, 2023-05-14:
In https://github.com/RainerKuemmerle/g2o/pull/685 I fixed the slam2d example for the master branch. It shows how to pass the sparse matrix for storage of the covariance and retrieving it again afterwards.

miquelmassot, 2023-05-15:
This is what I came up with:

```python 
from g2o import SparseBlockMatrixX

vertices = graph_slam.optimizer.vertices()

cov_vertices = []
covariances = SparseBlockMatrixX()
for vertex_idx in vertices:
    v = vertices[vertex_idx]
    if not v.fixed():
        cov_vertices.append((v.hessian_index(), v.hessian_index()))
print("Sparse pattern contains", len(cov_vertices), "blocks")

print("Computing covariance matrix...")
m = graph_slam.optimizer.compute_marginals(covariances, cov_vertices)
print("Done,", m)

print("Covariance matrix:")
for vertex_idx in vertices:
    print("Working on vertex idx:", vertex_idx)
    v = vertices[vertex_idx]
    if not v.fixed():
        print("Hessian idx is:", v.hessian_index())
        covariance = covariances.block(v.hessian_index(), v.hessian_index());
        print("Covariance is:", covariance)
```

and the results I see are:
 1. The bool return of `compute_marginals` is `False` - is this correct?
 2. I can retrieve the hessian indexes, but any access to the covariances matrix segfaults the program.

miquelmassot, 2023-05-16:
Full working example (that is not capable to retrieve the covariance) here: https://github.com/miquelmassot/g2o-python-examples/blob/main/g2o%20covariance%20retrieval.ipynb

RainerKuemmerle, 2023-05-19:
[image]
I believe my drawing function is broken but extracting the values seems to work.
I wonder what function was called beforehand. The `compute_marginals` was not wrapped on my pymem branch.
I pushed the branch I have been working on so far.

RainerKuemmerle, 2023-05-20:
[image]
Ok, this looks now better after rotating the covariance in a more proper way.

RainerKuemmerle, 2023-05-21:
@miquelmassot I updated https://github.com/RainerKuemmerle/g2o/blob/pymem/python/examples/notebook_slam2d.ipynb accordingly to plot the covariance of a PoseGraph.

miquelmassot, 2023-05-26:
Thanks, @RainerKuemmerle that's brilliant!
I have tried running your notebook with my g2o-python and am running into issues related to how the library is packaged in Python (PyPI) - so I need to solve that first and then I'll generate your same example on my side too.

With the current setup, I get the following error when loading the `manhattanOlson3500.g2o` file

```
virtual bool g2o::OptimizableGraph::load(std::istream&) unknown type: VERTEX_SE2
virtual bool g2o::OptimizableGraph::load(std::istream&) unknown type: EDGE_SE2
```

RainerKuemmerle, 2023-05-26:
For me the loading by the factory works because I link all types to the g2opy SO. Is this different in the g2o-python build?
I noticed that the types factory itself is also not wrapped to python which means we cannot query the known types in advance to better understand the error.

miquelmassot, 2023-06-23:
I am calling the same CMakeLists to wrap g2opy as g2o-python, so there shouldn't be any difference, but there is. I am still trying to find out why.

RainerKuemmerle, 2023-06-25:
Ha, I referenced a wrong issue in the above commit :(

@miquelmassot I added also some wrapper for the factory itself. Does this report empty lists for you?
https://github.com/RainerKuemmerle/g2o/blob/pymem/python/core/py_factory.cpp#L23

closed 2025-01-26 by github-actions[bot]