RainerKuemmerle/g2o #671: Is there a tool that can generate g2o edges and vertices?

w8128835, 2023-03-31:
hello,I want to test my algorithm to solve pose_graph problem and loop closure optimization,but i can't make a test case easil.
A long time ago a i used a tool named generate_random_graph.py,but i can not find it anymore.Is there a tool that can generate g2o edges and vertices?

RainerKuemmerle, 2023-03-31:
You can try g2o_simulator2d and g2o_simulator3d from this repository.
```
$ bin/g2o_simulator2d -h
Usage: bin/g2o_simulator2d [options] [graph-output]

General options:
-------------------------------------------
-help / -h           Displays this help.

Program Options:
-------------------------------------------
-hasCompass                  the robot has a compass
-hasGPS                      the robot has a GPS
-hasOdom                     the robot has an odometry
-hasPointBearingSensor       the robot has a point bearing sensor
-hasPointSensor              the robot has a point sensor
-hasPoseSensor               the robot has a pose sensor
-hasSegmentSensor            the robot has a segment sensor
-maxSegmentLength <double>   maximal Length of a segment in the world (default: 3)
-minSegmentLength <double>   minimal Length of a segment in the world (default: 0.5)
-nSegments <int>             number of segments (default: 1000)
-nlandmarks <int>            number of landmarks in the map (default: 100)
-segmentGridSize <int>       number of cells of the grid where to align the segments (default: 50)
-simSteps <int>              number of simulation steps (default: 100)
-worldSize <double>          size of the world (default: 25)
```

w8128835, 2023-03-31:
I got it ,thank you very much!

closed 2023-03-31 by w8128835