RainerKuemmerle/g2o #677: Unused SSE arguments warnings when compiling for ARM Android ABIs

TimPushkin, 2023-04-18:
When I cross-compile g2o for Android ABIs armeabi-v7a and arm64-v8a on my Windows 11 x86_64 machine I get the following warnings:
```
C/C++: clang++: warning: argument unused during compilation: '-msse2' [-Wunused-command-line-argument]
C/C++: clang++: warning: argument unused during compilation: '-msse3' [-Wunused-command-line-argument]
C/C++: clang++: warning: argument unused during compilation: '-msse4.1' [-Wunused-command-line-argument]
C/C++: clang++: warning: argument unused during compilation: '-msse4.2' [-Wunused-command-line-argument]
C/C++: clang++: warning: argument unused during compilation: '-msse4a' [-Wunused-command-line-argument]
```

felixendres, 2023-04-19:
That should be fine, as these SIMD extensions aren't available on ARM platforms, but probably [the autodetection in CMakeLists.txt](https://github.com/RainerKuemmerle/g2o/blob/master/CMakeLists.txt#L268-L303) gets that wrong.

TimPushkin, 2023-04-19:
Yeah, I understand that these aren't available on ARMs. It would be better if g2o wouldn't even try to use these flags on ARM to avoid generating these warnings.

RainerKuemmerle, 2023-05-01:
What if you set the option `DO_SSE_AUTODETECT` to `OFF`?
I guess you can do that when setting up your cross compiler?

TimPushkin, 2023-06-03:
I could, but there seems to beno easy way to set CMake flags for particular ABIs from Android Gradle Plugin (since I only need to disable autodetection for ARMs and not for x86). And I think it's an autodetection issue anyway, because if one needs to manually specify on what platforms to enable SSE and on what platforms to disable it, this doesn't sound like SSE autodetection :)

RainerKuemmerle, 2023-06-25:
I noticed a flaw how we use `check_cxx_compiler_flag`. The variable used for storing the result is cached and we used the same for all. Additionally, I tried to improve the check when using clang. Clang has a warning `unused-command-line-argument` which we can escalate to error. I pushed a commit but I had no chance to test with a Clang-based ARM toolchain.
@TimPushkin would be interesting how the branch works for you.

TimPushkin, 2023-06-25:
Yes, I don't get any `unused-command-line-argument` warnings on ARMs when using this branch. Thanks!

closed 2023-06-26 by RainerKuemmerle