RainerKuemmerle/g2o #682: "spdlog::spdlog" target missing when including g2o in another cmake project

matlabbe, 2023-05-12:
See https://github.com/introlab/rtabmap/issues/1031

It seems that this pull request https://github.com/RainerKuemmerle/g2o/pull/678 added spdlog logging ON by default. As the included target is public: 
```cmake
if (G2O_HAVE_LOGGING)
  if(TARGET spdlog::spdlog)
    target_link_libraries(stuff PUBLIC spdlog::spdlog)
  else()
    target_link_libraries(stuff PUBLIC spdlog::spdlog_header_only)
  endif()
endif()
```

The https://github.com/RainerKuemmerle/g2o/blob/master/cmake_modules/Config.cmake.in could have 
```cmake
if(@G2O_HAVE_LOGGING@)
  find_dependency(spdlog)
endif()
```
to let downstream packages including spdlog target.

RainerKuemmerle, 2023-05-13:
Thanks. I added a commit with your suggested fix.

closed 2023-05-13 by RainerKuemmerle

matlabbe, 2023-05-14:
Great! thx

PR #683 (2023-05-13): Add spdlog to g2o's exported config
Fix #682
commit d132be91 (2023-05-13): Add spdlog to g2o's exported config (#683)

Fix #682