RainerKuemmerle/g2o #688: Worse results after running PGO

VladimirYugay, 2023-05-24:
I have 4 trajectories: 2 predicted and 2 corresponding ground truths.

 2 predicted trajectories and 2 ground truths are concatenated and shown here: 

[image]

Now I want to improve the predicted trajectories. I define the correspondences between them and take respective ground truth transformations between the correspondences and add them as constraints. The correspondences are shown here:

[image]

Further, I run PGO (Python wrapper) with a very standard setup from this [readme](https://github.com/uoip/g2opy).

The optimized predicted trajectory is actually worse than the predicted one:

[image]

I also compute the metrics and they show that is actually worse not only visually. 

Is there something that I'm missing?

RainerKuemmerle, 2023-05-27:
Would it be possible to provide the g2o files of this examples?
Could it be that there is some "global" rotation in the data. Optimization fixes one node, for example the first one, and if the rotation has an error, the relative errors will on average be smaller but not necessarily align to the global poses of the ground truth.
Is your metric absolute or relative?
To handle that, consider aligning the trajectories with each other before computing an absolute error.

For python, maybe you want to try https://github.com/miquelmassot/g2o-python for a more up to date wrapper.

VladimirYugay, 2023-05-30:
Sure. See the file with the trajectories attached. 
[trajectories.zip](https://github.com/RainerKuemmerle/g2o/files/11596598/trajectories.zip)

I don't think that there's a global rotation - both trajectories are quite close to each other in the 3D space. See the visualization attached. I know it's not the best one, but the blue pyramids stand for the GT trajectory, and the orange one for the predicted one.

[image]

I measure the absolute translation error. Will try aligning them, thanks. 

Just to mention, I've tried the approach proposed by [PyPose](https://pypose.org/) from CVPR2023 and it optimized the trajectories almost perfectly given the same g2o file:

[image]

However, it's way more slower compared to g2o.

RainerKuemmerle, 2023-06-08:
The file `trajectories.g2o` is the optimized g2o result?

I noticed that you set two vertices to fixed. Both will not be changed during the optimization but used as anchors.
If their initial rotation is not how you expect them on the final result you might get some sort of rotated trajectories.
From your initial description I would assume that just one pose needs to fixed.

VladimirYugay, 2023-06-08:
Yes, it is an optimized g2o result. 

Here are the trajectories before optimization:  [raw_trajectories.zip](https://github.com/RainerKuemmerle/g2o/files/11687434/raw_trajectories.zip)

The fixed vertices are fixed by design: we don't want to change them during the optimization.

RainerKuemmerle, 2023-06-17:
You mentioned that PyPose yields a different result.
Do you know the chi² value of that solution?
raw/initial 0.122485
optimized 0.000484

If the solution of PyPose is lower than Gauss-Newton and Levenberg are running into a local minima.

closed 2025-01-26 by github-actions[bot]