RainerKuemmerle/g2o #689: How to use multithreads for g2o optimization

gitouni, 2023-05-26:
I tried to use multiple threadings for optimizing a huge Graph with almost 10M edges using `optimizer.initMultiThreading();` but failed.
# Cpp
```cpp
#include <g2o/core/solver.h>
#include <g2o/core/sparse_optimizer.h>
#include <g2o/core/optimization_algorithm_levenberg.h>
#include <g2o/solvers/cholmod/linear_solver_cholmod.h>
#include <g2o/core/optimization_algorithm_levenberg.h>

typedef g2o::BlockSolver<g2o::BlockSolverTraits<6, 3>> BlockSolverType; // This is not necessary unless Schur Complement is used
typedef g2o::LinearSolverCholmod<BlockSolverType::PoseMatrixType> LinearSolverType; // Linear Solver Type
auto solver = new g2o::OptimizationAlgorithmLevenberg(
        g2o::make_unique<BlockSolverType>(g2o::make_unique<LinearSolverType>()));
g2o::SparseOptimizer optimizer;    
optimizer.setAlgorithm(solver);  
optimizer.setVerbose(true);       
bool multithread_flag = optimizer.initMultiThreading();
    if(!multithread_flag)
        std::cerr << "use multithread for g2o failed!" << std::endl;
```
# Output
```
iteration= 0     chi2= 30318101608.807838        time= 7.83869   cumTime= 7.83869        edges= 11240810         schur= 0        lambda= 505254923932006.375000       levenbergIter= 1
iteration= 1     chi2= 29894616256.243656        time= 7.61399   cumTime= 15.4527        edges= 11240810         schur= 0        lambda= 266689971463425.968750       levenbergIter= 1
iteration= 2     chi2= 29728180818.956650        time= 7.65197   cumTime= 23.1046        edges= 11240810         schur= 0        lambda= 88896657154475.312500        levenbergIter= 1
iteration= 3     chi2= 29615177141.008823        time= 7.75045   cumTime= 30.8551        edges= 11240810         schur= 0        lambda= 29632219051491.769531        levenbergIter= 1
iteration= 4     chi2= 29613793696.719158        time= 9.56608   cumTime= 40.4212        edges= 11240810         schur= 0        lambda= 158038501607956.093750       levenbergIter= 3
```
The optimization time has not been reduced compared to one thread, and there has only one full-loaded CPU process when I use `htop` to check.
I truely appreciate your reply.

RainerKuemmerle, 2023-05-27:
You need to compile g2o with OpenMP support and have a BLAS which runs with multiple threads as well to support CHOLMOD's capabilities to factorize/solve multithreaded.

On Debian/Ubuntu, this should be possible using OpenBLAS.
1. configure g2o with G2O_USE_OPENMP = True
      - for this to work you might need to install the OpenMP libraries for your compiler
      - I am using clang16, so for me this is libomp-16-dev
      - I did not test/try with gcc
2. install openblas
      - on Debian/Ubuntu, it should become the default blas managed by update-alternatives
3. re-compile g2o

Example:
`~/workspace/g2o/build/bin/g2o -solver lm_fix6_3_cholmod -v -i 10 venice871.g2o`

htop (on a 6 cores AMD laptop):
[image]

gitouni, 2023-05-29:
Many thanks for your help @RainerKuemmerle , it works.

closed 2023-05-29 by gitouni