RainerKuemmerle/g2o #696: compile g2o error

qingchu123, 2023-06-25:
I have cloned the code from github and I compile the code with `cmake .. && make -j16`,then I have get the result:
```/home/fuyx/g2o/g2o/stuff/logger.cpp: In constructor ‘g2o::internal::LoggerInterface::LoggerInterface()’:
/home/fuyx/g2o/g2o/stuff/logger.cpp:42:13: error: ‘using element_type = class spdlog::logger’ {aka ‘class spdlog::logger’} has no member named ‘set_pattern’; did you mean ‘set_formatter’?
   42 |   console_->set_pattern("%+");
      |             ^~~~~~~~~~~
      |             set_formatter
/home/fuyx/g2o/g2o/stuff/logger.cpp: In destructor ‘g2o::internal::LoggerInterface::~LoggerInterface()’:
/home/fuyx/g2o/g2o/stuff/logger.cpp:45:47: error: ‘drop’ is not a member of ‘spdlog’
   45 | LoggerInterface::~LoggerInterface() { spdlog::drop("g2o"); }
      |                                               ^~~~
/home/fuyx/g2o/g2o/stuff/logger.cpp: In lambda function:
/home/fuyx/g2o/g2o/stuff/logger.cpp:67:5: error: ‘assert’ was not declared in this scope
   67 |     assert(false && "Unexpected level passed to the function");
      |     ^~~~~~
/home/fuyx/g2o/g2o/stuff/logger.cpp:36:1: note: ‘assert’ is defined in header ‘<cassert>’; did you forget to ‘#include <cassert>’?
   35 | #include <spdlog/sinks/stdout_color_sinks.h>
  +++ |+#include <cassert>
   36 | 
make[2]: *** [g2o/stuff/CMakeFiles/stuff.dir/build.make:202：g2o/stuff/CMakeFiles/stuff.dir/logger.cpp.o] 错误 1
make[1]: *** [CMakeFiles/Makefile2:1106：g2o/stuff/CMakeFiles/stuff.dir/all] 错误 2
make: *** [Makefile:156：all] 错误 2
```
so what should i do for this?

RainerKuemmerle, 2023-06-25:
Which version of spdlog do you have installed?
You can also disable logging by setting G2O_USE_LOGGING to false using, e.g., cmake-gui in your build folder.

qingchu123, 2023-06-25:
i am using the last spdlog by clonning the spdlog's github address and use `cmake ..&& make -j16&&sudo make install` to install @RainerKuemmerle

RainerKuemmerle, 2023-06-25:
https://github.com/gabime/spdlog/blob/v1.x/include/spdlog/logger.h#L326
The method exists on the v1.x branch. I'd recommend the package of your distribution or a release tag.
I am still not sure which version you are actually using.

RainerKuemmerle, 2023-11-12:
No further information provided

closed 2023-11-12 by RainerKuemmerle