RainerKuemmerle/g2o #699: g2o file format EDGE_SE3:QUAT information matrix

XihangYU630, 2023-06-26:
One question: Does anyone know in g2o pose graph solver, it processes .g2o file format EDGE_SE3:QUAT information matrix as covariance of [x, y, z, roll, pitch, yaw] (Euler Angle), [x, y, z, qx, qy, qz] (quaternion) or other formats? I cannot find any official documentation here. Thanks!

RainerKuemmerle, 2023-06-29:
It's  [x, y, z, qx, qy, qz]

anilesec, 2023-07-17:
@RainerKuemmerle 

I have difficulty in computing the information matrix for the following example(I have vertex and edge with constraints, but I don't know how is information matrix is obtained). Could you please show one simple example? sorry if its too simple question, I am new to this SLAM 

VERTEX_SE3:QUAT 0 0.000000 0.000000 0.000000 0.0000000 0.0000000 0.0000000 1.0000000
VERTEX_SE3:QUAT 1 0.939375 -0.022347 -0.028831 0.1175925 -0.3414281 -0.7976288 0.4831016

EDGE_SE3:QUAT 0 1   0.939375 -0.022347 -0.028831   0.1175925 -0.3414281 -0.7976288 0.4831016   100.000000 0.000000 0.000000 0.000000 0.000000 0.000000   100.000000 0.000000 0.000000 0.000000 0.000000   100.000000 0.000000 0.000000 0.000000   400.000000 0.000000 0.000000   400.000000 0.000000   400.000000

Can you please help me understand how the information matrix compute?

Thanks in advance!

closed 2025-01-25 by github-actions[bot]