RainerKuemmerle/g2o #704: Geman McClure robust kernel

OsamaAlAhmadieh, 2023-07-18:
Hello, 

If I understood correctly, in the robust kernel `robustify` function, we pass the squared error, and return the loss term, its first derivative and its second derivative. The derivatives are taken w.r.t $e^2$ where $e$ is the error. 
For the Geman McClure function the formula is 
$$\frac{c^2 e^2}{c^2 + e^2}$$
In the [Implementation](https://github.com/RainerKuemmerle/g2o/blob/672aa7a2f984553231283055a3bfb0f8c2757765/g2o/core/robust_kernel_impl.cpp#L95C6-L95C41) the `_delta` corresponds to $c^2$ and `e2` corresponds to $e^2$. 
The loss `rho[0]` is then calculated as `e2 * aux` where `aux = _delta / _delta + e2`.  
The first derivate of the loss w.r.t $e^2$ is $$\frac{c^4}{(c^2 + e^2)^2}$$ and is implemented as `aux * aux`
The second derivative of the loss w.r.t $e^2$ is $$\frac{-2c^4}{(c^2 + e^2)^3}$$ But this in the code is implemented as `rho[2] =  -2. * rho[1] * aux` which means that there is an extra $c^2$ or `_delta` in the numerator. 
I am not sure if this would heavily affect the optimization though. I have found the literature for the GM online a bit not consistent, but I checked based on the [provided thesis, section 5.4.2](http://www2.informatik.uni-freiburg.de/~agarwal/resources/agarwal-thesis.pdf).
Thank you

RainerKuemmerle, 2023-07-23:
Thank you very much for spotting this. I took the chance to implement a quick unit test which evaluates the second derivative numerically by finite difference. I found that the implementation of another kernel was wrong.

OsamaAlAhmadieh, 2023-07-25:
Hello,
thanks for fixing it !

commit 0f70abff (2023-07-29): Fix Geman McClure robust kernel

Fix #704
commit d3d94d79 (2023-07-29): Fix Geman McClure robust kernel

Fix #704
closed 2023-07-29 by RainerKuemmerle