RainerKuemmerle/g2o #708: how to run the benchmark?

xshen053, 2023-08-02:
I set flag in CMakeLists of benchmark to ON, but looks like it cannot find it.
```
CMake Warning at CMakeLists.txt:532 (find_package):
  By not providing "Findbenchmark.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "benchmark", but CMake did not find one.

  Could not find a package configuration file provided by "benchmark" with
  any of the following names:

    benchmarkConfig.cmake
    benchmark-config.cmake

  Add the installation prefix of "benchmark" to CMAKE_PREFIX_PATH or set
  "benchmark_DIR" to a directory containing one of the above files.  If
  "benchmark" provides a separate development package or SDK, be sure it has
  been installed.

CMake Warning at CMakeLists.txt:536 (message):
  G2O_BUILD_BENCHMARKSwassettotrue,butthebenchmarklibrarycannotbefound
```

xshen053, 2023-08-02:
Solved, reason is I didn't install google benchmark

RainerKuemmerle, 2023-08-06:
The benchmark focus on a very particular aspect on how to allocate an array.
Might not provide much insights into things like which linear solver is faster depending on the size of problem or other aspects.
I will fix the benchmark that it can be compiled.

PR #709 (2023-08-06): Fix compile errors for the benchmarks
Fix #708
commit 2e050e42 (2023-08-06): Fix compile errors for the benchmarks

Fix #708
closed 2023-08-06 by RainerKuemmerle