RainerKuemmerle/g2o #711: Tests fail to compile: RainerKuemmerle

yurivict, 2023-08-08:
```
ld: error: undefined symbol: testing::AssertionSuccess()
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(Data_DataQueue_Test::TestBody())
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(Data_DataQueue_Test::TestBody())
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(testing::AssertionResult testing::internal::CmpHelperFloatingPointEQ<double>(char const*, char const*, double, double))
>>> referenced 3 more times

ld: error: undefined symbol: testing::Message::Message()
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(Data_DataQueue_Test::TestBody())
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(Data_DataQueue_Test::TestBody())
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(Data_DataQueue_Test::TestBody())
>>> referenced 13 more times

ld: error: undefined symbol: testing::internal::AssertHelper::AssertHelper(testing::TestPartResult::Type, char const*, int, char const*)
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(Data_DataQueue_Test::TestBody())
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(Data_DataQueue_Test::TestBody())
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(Data_DataQueue_Test::TestBody())
>>> referenced 12 more times

ld: error: undefined symbol: testing::internal::AssertHelper::operator=(testing::Message const&) const
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(Data_DataQueue_Test::TestBody())
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(Data_DataQueue_Test::TestBody())
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(Data_DataQueue_Test::TestBody())
>>> referenced 12 more times

ld: error: undefined symbol: testing::internal::AssertHelper::~AssertHelper()
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(Data_DataQueue_Test::TestBody())
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(Data_DataQueue_Test::TestBody())
>>> referenced by io_data.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/io_data.cpp.o:(Data_ReadWriteRobotLaser_Test::TestBody())
>>> referenced 7 more times

ld: error: undefined symbol: testing::internal::GetTestTypeId()
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(_GLOBAL__sub_I_data_queue_tests.cpp)
>>> referenced by io_data.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/io_data.cpp.o:(_GLOBAL__sub_I_io_data.cpp)

ld: error: undefined symbol: testing::internal::MakeAndRegisterTestInfo(char const*, char const*, char const*, char const*, testing::internal::CodeLocation, void const*, void (*)(), void (*)(), testing::internal::TestFactoryBase*)
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(_GLOBAL__sub_I_data_queue_tests.cpp)
>>> referenced by io_data.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/io_data.cpp.o:(_GLOBAL__sub_I_io_data.cpp)

ld: error: undefined symbol: testing::internal::IsTrue(bool)
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(testing::internal::SuiteApiResolver<testing::Test>::GetSetUpCaseOrSuite(char const*, int))
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(testing::internal::SuiteApiResolver<testing::Test>::GetTearDownCaseOrSuite(char const*, int))
>>> referenced by io_data.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/io_data.cpp.o:(testing::AssertionResult testing::internal::PredicateFormatterFromMatcher<testing::internal::SizeIsMatcher<unsigned long>>::operator()<std::__1::vector<double, std::__1::allocator<double>>>(char const*, std::__1::vector<double, std::__1::allocator<double>> const&) const)
>>> referenced 27 more times

ld: error: undefined symbol: testing::internal::GTestLog::GTestLog(testing::internal::GTestLogSeverity, char const*, int)
>>> referenced by data_queue_tests.cpp
>>>               unit_test/data/CMakeFiles/unittest_data.dir/data_queue_tests.cpp.o:(testing::internal::SuiteApiResolver<testing::Test>::GetSetUpCaseOrSuite(char const*, int))
[... 245 lines cut ...]

ld: error: undefined symbol: typeinfo for testing::Test
>>> referenced by auto_diff.cpp
>>>               unit_test/general/CMakeFiles/unittest_general.dir/auto_diff.cpp.o:(typeinfo for AutoDifferentiation)
>>> referenced by auto_diff.cpp
>>>               unit_test/general/CMakeFiles/unittest_general.dir/auto_diff.cpp.o:(typeinfo for AutoDifferentiationEdgeSE2)
>>> referenced by batch_statistics.cpp
>>>               unit_test/general/CMakeFiles/unittest_general.dir/batch_statistics.cpp.o:(typeinfo for General_BatchStatisticsOutput_Test)
>>> referenced 28 more times

ld: error: undefined symbol: testing::internal::GetTestTypeId()
>>> referenced by batch_statistics.cpp
>>>               unit_test/general/CMakeFiles/unittest_general.dir/batch_statistics.cpp.o:(_GLOBAL__sub_I_batch_statistics.cpp)
>>> referenced by graph_operations.cpp
>>>               unit_test/general/CMakeFiles/unittest_general.dir/graph_operations.cpp.o:(_GLOBAL__sub_I_graph_operations.cpp)
>>> referenced by graph_operations.cpp
>>>               unit_test/general/CMakeFiles/unittest_general.dir/graph_operations.cpp.o:(_GLOBAL__sub_I_graph_operations.cpp)
>>> referenced 16 more times

ld: error: too many errors emitted, stopping now (use --error-limit=0 to see all errors)
```

Version: 20230806
clang-14
FreeBSD 13.2

RainerKuemmerle, 2023-08-12:
I need more information here.
Using a VirtualBox machine (freebsd 13.2 amd64)
```
$ clang --version
FreeBSD clang version 14.0.5 (https://github.com/llvm/llvm-project.git llvmorg-14.0.5-0-gc12386ae247c)
Target: x86_64-unknown-freebsd13.2
Thread model: posix
InstalledDir: /usr/bin
```
I can nicely build, for example, the stuff test is afterwards linked as follows to gtest
```$ ldd bin/unittest_stuff 
bin/unittest_stuff:
	libg2o_stuff.so.0.1 => /home/goki/workspace/g2o/build/lib/libg2o_stuff.so.0.1 (0x3e6bc17d2000)
	libgmock_main.so.1.14.0 => /home/goki/workspace/g2o/build/lib/libgmock_main.so.1.14.0 (0x3e6bc2f9f000)
	libgmock.so.1.14.0 => /home/goki/workspace/g2o/build/lib/libgmock.so.1.14.0 (0x3e6bc19a2000)
	libgtest.so.1.14.0 => /home/goki/workspace/g2o/build/lib/libgtest.so.1.14.0 (0x3e6bc1aee000)
	libthr.so.3 => /lib/libthr.so.3 (0x3e6bc202d000)
	libc++.so.1 => /usr/lib/libc++.so.1 (0x3e6bc3d66000)
	libcxxrt.so.1 => /lib/libcxxrt.so.1 (0x3e6bc4cb7000)
	libm.so.5 => /lib/libm.so.5 (0x3e6bc4ce2000)
	libgcc_s.so.1 => /lib/libgcc_s.so.1 (0x3e6bc541b000)
	libc.so.7 => /lib/libc.so.7 (0x3e6bc6100000)
	[vdso] (0x7ffffffff5d0)
```
Please provide the output of `make VERBOSE=1` or more information on how your linker is called. Or which command sequence you used.

I found a dependency issue problem in one of the examples when there is no OPENGL found but that should be unrelated to the linking issue for gtest.

RainerKuemmerle, 2023-08-19:
@yurivict Would be interesting to understand if master fixes the issue for you. Thanks.

closed 2025-01-25 by github-actions[bot]