RainerKuemmerle/g2o #714: Fails on Compilation with this error code

chetanpandey1266, 2023-08-15:
```shell
 94%] Built target simple_optimize
[ 95%] Linking CXX executable ../../../bin/convertSegmentLine
/usr/bin/ld: cannot find -lg2o_simulator_library: No such file or directory
collect2: error: ld returned 1 exit status
make[2]: *** [g2o/examples/convert_segment_line/CMakeFiles/convertSegmentLine_application.dir/build.make:102: bin/convertSegmentLine] Error 1
make[1]: *** [CMakeFiles/Makefile2:2218: g2o/examples/convert_segment_line/CMakeFiles/convertSegmentLine_application.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

RainerKuemmerle, 2023-08-15:
Which version do you try to compile. I pushed https://github.com/RainerKuemmerle/g2o/commit/871e0e8bad39e7475493d556456774a7d0287955 to master which relates to the issue. Does the master branch work for you?

chetanpandey1266, 2023-08-15:
How should I build directly from the master branch?

RainerKuemmerle, 2023-08-15:
```
git clone https://github.com/RainerKuemmerle/g2o.git
```
Will get you the code via git and you can try the master branch.

closed 2023-11-12 by RainerKuemmerle