RainerKuemmerle/g2o #719: g2o::make_unique not recognized

NadimDeeb, 2023-08-31:
I am building my ros2 workspace which includes the lidarslam_ros2 package which uses the g2o library. When I colcon build I get the following error which tells me that the make_unique is not a member of g2o and them suggests me to use the std::make_unique. Here is the error exactly:

`/home/nadimdib/ros2_3d/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp:272:10: error: ‘make_unique’ is not a member of ‘g2o’; did you mean ‘std::make_unique’?
  272 |     g2o::make_unique<g2o::LinearSolverEigen<g2o::BlockSolver_6_3::PoseMatrixType>>();
In file included from /usr/include/c++/9/memory:80,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:144,
                 from /home/nadimdib/ros2_3d/src/lidarslam_ros2/graph_based_slam/include/graph_based_slam/graph_based_slam_component.h:42,
                 from /home/nadimdib/ros2_3d/src/lidarslam_ros2/graph_based_slam/src/graph_based_slam_component.cpp:1:
/usr/include/c++/9/bits/unique_ptr.h:868:5: note: ‘std::make_unique’ declared here
  868 |     make_unique(_Args&&...) = delete;
`

headlee, 2023-09-13:
I ran into this same issue when trying to build lidarslam with a new version of g2o. It looks like [this commit](https://github.com/RainerKuemmerle/g2o/commit/6888d9fe9274e23678427c6ece09f92e40c71544) back in March dropped `g2o::make_unique`. I believe your options are to:

1. Use an older version of g2o. Not sure how easy this is on Ubuntu. I'm building from source on Windows and older versions of g2o aren't finding my Eigen3, so I decided to go with option 2 below instead.

2. Update lidarslam to use `std::make_unique` (change lines 272 and 274 in [this file](https://github.com/rsasaki0109/lidarslam_ros2/blob/b657c7be2079b441952177550b24562a7fc5e05f/graph_based_slam/src/graph_based_slam_component.cpp#L272))

I'm able to correctly build on Windows after option 2.

RainerKuemmerle, 2023-09-17:
Yes, after moving to c++17 as requirement the helper which was in `misc.h` is no longer needed. Changing to use `std::make_unique` is the correct approach.

closed 2023-11-12 by RainerKuemmerle