RainerKuemmerle/g2o #721: Linker error: multiple definition of `typeinfo name for g2o::BaseVertex<7, g2o::Sim3>';

dkogan, 2023-10-03:
Hello. This is the inverse of #444. I just pushed g2o to Debian. On armhf there's a build error:

https://buildd.debian.org/status/fetch.php?pkg=g2o&arch=armhf&ver=0%7E20230806-1&stamp=1696268512&raw=0

Look at the end. Removing the fix for #444 makes the build complete, at least on armhf. Do you have a clear understanding of #444, and what this failure means? Should the template be explicitly instantiated, or not?

Thanks

wangzheqie, 2023-10-03:
这是来自QQ邮箱的自动回复邮件。你好，我已经收到你的邮件，尽快处理。

RainerKuemmerle, 2023-10-20:
@dkogan Do you know if removing this line would work on other architectures?

Back then I thought it passes in general but avoids an issue on MSVC. But MSVC is somewhat not so clear to me and we already have a few `#ifdef MSC_VER` in the code.

dkogan, 2023-10-20:
I currently have a distro-patch to remove that line, and here's what the build results are: https://buildd.debian.org/status/package.php?p=g2o

The i386 and arch=all failures are unrelated. So I would say that as far as Debian is concerned, removing that line is the right thing to do.

PR #727 (2023-10-21): Add workaround for multiple definitions
Fix #721
Relates #444
commit a601119b (2023-10-21): Add workaround for multiple definitions

Fix #721
Relates #444
closed 2023-10-21 by RainerKuemmerle