RainerKuemmerle/g2o #722: Tests fail on i386

dkogan, 2023-10-07:
Hello. The Debian g2o package is failing on i386. The build log: https://buildd.debian.org/status/fetch.php?pkg=g2o&arch=i386&ver=0%7E20230806-2&stamp=1696420212&raw=0

It looks like there's a test failure:

```
./unit_test/slam3d/optimization_slam3d.cpp:94: Failure
Value of: v2AfterOpti->estimate().translation().isApprox(g2o::Vector3::Zero())
  Actual: false
Expected: true
```

Can we please improve the error message? I suspect that this is simply an issue of numerical fuzz, and increasing the threshold used by `isApprox()` would fix this. It would be nice if the error message clearly said what threshold it's using, and what value it saw instead.

Thanks

RainerKuemmerle, 2023-10-20:
Agree, we should use test code that better prints actual and expected value.
Will look at this, in particular, because on https://eigen.tuxfamily.org/dox/classEigen_1_1DenseBase.html#ae8443357b808cd393be1b51974213f9c I found the interesting note that it does not work to check for approximately zero.

RainerKuemmerle, 2023-10-21:
Pushed a commit with which a failing test would print, e.g.,
```
/home/goki/workspace/g2o/unit_test/slam3d/optimization_slam3d.cpp:139: Failure
Value of: print_wrap(v2AfterOpti->estimate().rotation().diagonal())
Expected: does differ from { 0, 0, 0 }
more than 0.001
  Actual: { 0.99999999999999989, 0.99999999999999989, 0.99999999999999989 }
```

dkogan, 2023-10-30:
Hi. I cherry-picked that patch into the g2o package, and it builds on i386 now. To confirm: that patch adds extra diagnostics AND loosens the acceptance threshold? Yes? Thanks for patching this. I'm closing the bug.

closed 2023-10-30 by dkogan

RainerKuemmerle, 2023-11-01:
Yes. With the patch we add diagnostics and an adapted acceptance threshold for the test.