RainerKuemmerle/g2o #724: undefined reference to `vtable for g2o::VertexPointXYZ'

DizkiBorsuk, 2023-10-14:
Hi, I'm trying to use g2o::VertexPointXYZ but i'm getting this error: 

warning: relocation against `_ZTVN3g2o14VertexPointXYZE' in read-only section `.text._ZN3g2o14VertexPointXYZC2Ev[_ZN3g2o14VertexPointXYZC5Ev]'
[build] ...: in function `g2o::VertexPointXYZ::VertexPointXYZ()':
[build] /usr/local/include/g2o/types/slam3d/vertex_pointxyz.h:41: undefined reference to `vtable for g2o::VertexPointXYZ'
[build] /usr/bin/ld: /usr/local/include/g2o/types/slam3d/vertex_pointxyz.h:41: undefined reference to `vtable for g2o::VertexPointXYZ'
[build] /usr/bin/ld: warning: creating DT_TEXTREL in a PIE

Thanks for help in advance

rohanx01, 2023-10-17:
make sure vertex_pointxyz.cpp is compiled simultaneously with your code.. for eg: use g++ vertex_pointxyz.cpp  your_file.cpp -o my_program 
then run ./my_program

DizkiBorsuk, 2023-10-18:
That worked, thank you

closed 2023-10-18 by DizkiBorsuk