RainerKuemmerle/g2o #732: RuntimeError: logger with name 'g2o' already exists

shrimo, 2023-11-04:
Hello
I'm trying to run the example **sba_demo.py**, the following error occurs: 

`Performing full BA:
Traceback (most recent call last):
  File "/g2o-pymem/python/examples/sba_demo.py", line 139, in <module>
    main()
  File "/home/cds/github/g2o-pymem/python/examples/sba_demo.py", line 120, in main
    optimizer.optimize(10)
RuntimeError: logger with name 'g2o' already exists
`

https://github.com/RainerKuemmerle/g2o/blob/pymem/python/examples/sba_demo.py#L120

sba_demo2.py - works without errors

Building a framework on Fedora Linux 38 for Python 3.10

`cmake -DBUILD_SHARED_LIBS=ON -DG2O_USE_OPENGL=OFF -DG2O_BUILD_EXAMPLES=OFF -DG2O_BUILD_APPS=OFF -DG2O_BUILD_PYTHON=ON -DPYBIND11_INSTALL=ON -DCMAKE_POSITION_INDEPENDENT_CODE=ON ../
`

I would be very grateful for any advice on this matter.

RainerKuemmerle, 2023-11-11:
Thanks. I added a workaround for the logger singleton. The "simple" singleton uses a static variable which is local to each library. Hence, we ended up adding multiple loggers.

shrimo, 2023-11-11:
> Thanks. I added a workaround for the logger singleton. The "simple" singleton uses a static variable which is local to each library. Hence, we ended up adding multiple loggers.

thank you very much for the answer

commit b90c05eb (2023-11-11): Fix logger singleton

Fix #732
PR #736 (2023-11-12): Fix logger singleton
Fix #732
commit 7023b80e (2023-11-12): Fix logger singleton (#736)

* Fix logger singleton
* Fix includes without G2O_HAVE_LOGGING

Fix #732
closed 2023-11-12 by RainerKuemmerle