RainerKuemmerle/g2o #735: When I build g2o, I meet error with spdlog

JimXu1989, 2023-11-08:
[  1%] Built target freeglut_minimal
[  1%] Linking CXX shared library ../../lib/libg2o_stuff.so
/usr/bin/ld: /usr/local/lib/libspdlog.a(spdlog.cpp.o): relocation R_X86_64_TPOFF32 against `_ZGVZN6spdlog7details2os9thread_idEvE3tid' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: failed to set dynamic section sizes: bad value
collect2: error: ld returned 1 exit status
make[2]: *** [g2o/stuff/CMakeFiles/stuff.dir/build.make:243: lib/libg2o_stuff.so.0.1.0] Error 1
make[1]: *** [CMakeFiles/Makefile2:1124: g2o/stuff/CMakeFiles/stuff.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

How can I solve this problem?

RainerKuemmerle, 2023-11-08:
Which distribution are you using? Why is there a static library of spdlog?
Did you build spdlog on your own, you might want to check https://github.com/gabime/spdlog/issues/1405

JimXu1989, 2023-11-11:
> Which distribution are you using? Why is there a static library of spdlog? Did you build spdlog on your own, you might want to check [gabime/spdlog#1405](https://github.com/gabime/spdlog/issues/1405)

Viele Dank! That's work!

closed 2023-11-11 by RainerKuemmerle