RainerKuemmerle/g2o #742: g2o issues with ROS2 humble .

chrissunny94, 2023-11-21:
```
undefined reference to symbol '_ZN3fmt2v86detail18throw_format_errorEPKc'
/usr/bin/ld: /lib/x86_64-linux-gnu/libfmt.so.8: error adding symbols: DSO missing from command line
```

RainerKuemmerle, 2023-11-21:
When are you getting this linker error? When you compile g2o or when you link your code with g2o?

chrissunny94, 2023-11-22:
While linking with g2o .
Getting it with the following .
https://github.com/Chris7462/floam/tree/g2o

Same code worked with ROS2 FOXY on Ubuntu 20.04 .

I have insalled g2o from the ROS2 release , `sudo apt install ros-<distro>-libg2o` , i am not sure of the corresponding g2o release with the same .

I will trying installing the **FOXY-bundled-g2o** from source and revert back for further clarity .

RainerKuemmerle, 2023-11-22:
I am not following the ROS2 releases in detail. But I assume a different release is used as the dependency to fmt/spdlog is rather new.How is linking performed? Are you using the target files exported by cmake. In ROS1 noetic this is tedious but prevents issues. Not sure about ROS2 again.

https://github.com/Chris7462/floam/tree/g2o

chrissunny94, 2023-11-23:
Noted! 

I will try to dig in over this weekend .

Thank you

closed 2025-01-25 by github-actions[bot]