RainerKuemmerle/g2o #749: Cannot link g2o to CMake project, even after building successfully from source and installing to system paths

JeffR1992, 2023-12-05:
I built and installed g2o from source, and calling
`which g2o`

shows me that the g2o binary is installed to:
`/usr/local/bin/g2o`

Furthermore, I can find the headers by going to:
`/usr/local/include/g2o`

which gives the following directories:
`EXTERNAL  apps  autodiff  config.h  core  examples  solvers  stuff  types`

However, in my project's CMakeLists.txt file, when I add
`find_package(G2O REQUIRED)`

I get the error:
```
CMake Error at /home/jeff/projects/catkin_ws/src/mapper/CMakeLists.txt:34 (find_package):
  By not providing "FindG2O.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "G2O", but
  CMake did not find one.

  Could not find a package configuration file provided by "G2O" with any of
  the following names:

    G2OConfig.cmake
    g2o-config.cmake

  Add the installation prefix of "G2O" to CMAKE_PREFIX_PATH or set "G2O_DIR"
  to a directory containing one of the above files.  If "G2O" provides a
  separate development package or SDK, be sure it has been installed.
```

I also tried copying over the `cmake_modules` directory from the g2o source directory into my project root directory and added the `cmake_modules` directory to the `CMAKE_MODULE_PATH` with:
`list(APPEND CMAKE_MODULE_PATH ${CMAKE_CURRENT_SOURCE_DIR}/cmake_modules)`

In this way, I don't get the above error anymore, but if I print out the g2o environment variables:
`message(STATUS "G2O_INCLUDE_DIR: ${G2O_INCLUDE_DIR}")`
`message(STATUS "G2O_LIBRARIES: ${G2O_LIBRARIES}")`

I see blank paths:
`G2O_INCLUDE_DIR:`
`G2O_LIBRARIES:`

Which causes linking errors if I try and actually use something like `g2o::SparseOptimizer optimizer` in my project.

So my question is, how in the world do I get CMake to see g2o so I can link my project to it correctly? I feel like what I've tried should have solved the issue, yet I've had no luck. Any help would be appreciated.

RainerKuemmerle, 2023-12-17:
There is a ROS package available in ROS1. Did you try using that instead of building from src? See for example, https://github.com/rst-tu-dortmund/teb_local_planner for a package using that.

geoeo, 2024-06-18:
This still seems to be an issue on master and is not related to ROS.  I cant link use this library when built work source.

apmilko, 2024-07-22:
@RainerKuemmerle can you please provide an answer to this issue?

closed 2025-01-25 by github-actions[bot]