RainerKuemmerle/g2o #752: How to clear g2o optimizer after optimization to avoid memory leak

nkuwenjian, 2023-12-21:
Hi, I have a following simple code, which is used to study the usage of g2o solver. This code works normally after compilation. However, when I use the tool of valgrind to check the code, I encounter the following error prompt about memory leak. So my question is how to properly clear g2o optimizer after optimization to avoid memory leak.

The version of g2o I used is the release version of 20200410.

The command of valgrind I used is `valgrind --tool=memcheck --leak-check=full ./g2o_tutorial`

Best,

Jian Wen

```
#include "g2o/core/block_solver.h"
#include "g2o/core/factory.h"
#include "g2o/core/optimization_algorithm_factory.h"
#include "g2o/core/optimization_algorithm_gauss_newton.h"
#include "g2o/solvers/cholmod/linear_solver_cholmod.h"
#include "g2o/types/slam3d/types_slam3d.h"

using BlockSolver = g2o::BlockSolver<g2o::BlockSolverTraits<-1, -1>>;
using LinearSolver = g2o::LinearSolverCholmod<BlockSolver::PoseMatrixType>;

int main() {
  // Initialize the SparseOptimizer
  g2o::SparseOptimizer optimizer;
  auto linear_solver = std::make_unique<LinearSolver>();
  linear_solver->setBlockOrdering(false);
  auto block_solver = std::make_unique<BlockSolver>(std::move(linear_solver));
  g2o::OptimizationAlgorithmGaussNewton* solver =
      new g2o::OptimizationAlgorithmGaussNewton(std::move(block_solver));
  optimizer.setAlgorithm(solver);

  // Add vertices
  g2o::VertexSE3* v0 = new g2o::VertexSE3();
  v0->setEstimate(
      Eigen::Transform<double, 3, 1>(Eigen::Translation<double, 3>(0, 0, 0)));
  v0->setId(0);
  optimizer.addVertex(v0);

  g2o::VertexSE3* v1 = new g2o::VertexSE3();
  v1->setEstimate(
      Eigen::Transform<double, 3, 1>(Eigen::Translation<double, 3>(0, 0, 0)));
  v1->setId(1);
  optimizer.addVertex(v1);

  g2o::VertexSE3* v2 = new g2o::VertexSE3();
  v2->setEstimate(
      Eigen::Transform<double, 3, 1>(Eigen::Translation<double, 3>(0, 0, 0)));
  v2->setId(2);
  optimizer.addVertex(v2);

  // Add edges
  g2o::EdgeSE3* e1 = new g2o::EdgeSE3();
  e1->vertices()[0] = optimizer.vertex(0);
  e1->vertices()[1] = optimizer.vertex(1);
  e1->setMeasurement(Eigen::Isometry3d(Eigen::Translation<double, 3>(1, 0, 0)));
  e1->setInformation(Eigen::Matrix<double, 6, 6>::Identity());
  optimizer.addEdge(e1);

  g2o::EdgeSE3* e2 = new g2o::EdgeSE3();
  e2->vertices()[0] = optimizer.vertex(1);
  e2->vertices()[1] = optimizer.vertex(2);
  e2->setMeasurement(Eigen::Isometry3d(Eigen::Translation<double, 3>(0, 1, 0)));
  e2->setInformation(Eigen::Matrix<double, 6, 6>::Identity());
  optimizer.addEdge(e2);

  g2o::EdgeSE3* e3 = new g2o::EdgeSE3();
  e3->vertices()[0] = optimizer.vertex(2);
  e3->vertices()[1] = optimizer.vertex(0);
  e3->setMeasurement(
      Eigen::Isometry3d(Eigen::Translation<double, 3>(-0.8, -0.7, 0.1)));
[... 191 lines cut ...]
==139659==    by 0x4011CA0: call_init (dl-init.c:30)
==139659==    by 0x4011CA0: _dl_init (dl-init.c:119)
==139659==    by 0x4001139: ??? (in /usr/lib/x86_64-linux-gnu/ld-2.31.so)
==139659== 
==139659== 120 bytes in 1 blocks are definitely lost in loss record 23 of 23
==139659==    at 0x483BE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==139659==    by 0x486C629: g2o::HyperGraphActionLibrary::registerAction(g2o::HyperGraphElementAction*) (in /usr/local/lib/libg2o_core.so)
==139659==    by 0x48F16ED: __static_initialization_and_destruction_0(int, int) [clone .constprop.955] (in /usr/local/lib/libg2o_types_slam3d.so)
==139659==    by 0x4011B99: call_init.part.0 (dl-init.c:72)
==139659==    by 0x4011CA0: call_init (dl-init.c:30)
==139659==    by 0x4011CA0: _dl_init (dl-init.c:119)
==139659==    by 0x4001139: ??? (in /usr/lib/x86_64-linux-gnu/ld-2.31.so)
==139659== 
==139659== LEAK SUMMARY:
==139659==    definitely lost: 512 bytes in 19 blocks
==139659==    indirectly lost: 0 bytes in 0 blocks
==139659==      possibly lost: 0 bytes in 0 blocks
==139659==    still reachable: 200 bytes in 4 blocks
==139659==         suppressed: 0 bytes in 0 blocks
==139659== Reachable blocks (those to which a pointer was found) are not shown.
==139659== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==139659== 
==139659== For lists of detected and suppressed errors, rerun with: -s
==139659== ERROR SUMMARY: 19 errors from 19 contexts (suppressed: 0 from 0)
```

RainerKuemmerle, 2023-12-21:
The "memory leak" originates from singletons which are not freed upon exit. The amount of bytes lost is independent of the optimization problem, its size or how often you run it.
Fixing this would require to change the singleton pattern implementation to correctly free its data.

RainerKuemmerle, 2024-02-18:
Fixed by #472 see also #148

closed 2024-02-18 by RainerKuemmerle

nkuwenjian, 2024-03-06:
Thank you. I use the release version of 20230223 instead of 20200410, and the problem is solved.