RainerKuemmerle/g2o #759: How can I print a variable in the ComputeError() in the terminal？

yangliu9527, 2024-01-25:
I want to print some values calculated in the computeError() function of an edge during the optimization process, but it seems that the output stream like std::cout/std::cerr doesn't work in the computeError(). I have tried the write(ostream &os) function but it is used for save information in a file. I want to see some information in real-time because there is a unknown bug which will cause the system crashing. Do you know the solution? Thank you.

RainerKuemmerle, 2024-02-01:
Do you have a minimal working example that illustrates the problem?
Using cout/cerr should work as expected. In the g2o_viewer cerr is redirected to the GUI and you should see the output there instead of on the console.

About the crash:
Does gdb provide insights? Or tools like valgrind?

yangliu9527, 2024-02-07:
void EdgeLaneAlignOnlyPose::computeError()
    {
        const g2o::VertexSE3Expmap *v1 = static_cast<const g2o::VertexSE3Expmap *>(_vertices[0]);
        Eigen::Vector2d proj_p = pCamera->project(v1->estimate().map(Pw));
        double proj_x = proj_p[0];
        double proj_y = proj_p[1];
        if (proj_x <= 1 || proj_x >= (pImEdgeDT->cols - 1) || proj_y <= 1 || proj_y >= (pImEdgeDT->rows - 1))
        {
            setLevel(1);
            _error.setZero();
            return;
        }
        else
        {
            // double error = static_cast<double>(pImEdgeDT->ptr<float>(proj_y)[proj_x]);
            double error = static_cast<double>(Utility::GetPixelValue(pImEdgeDT, proj_x, proj_y));
            cout << error <<endl;
            _error(0, 0) = error;
        }
    }

Here is an example of working computeError() function, where the cout << error << endl; doesn't work. Do you have any solution? Thank you.

RainerKuemmerle, 2024-02-10:
Try to use a debugger along with a breakpoint in the respective functions. Without a minimal working example, I cannot tell what is going on.

closed 2025-01-24 by github-actions[bot]