RainerKuemmerle/g2o #776: Same codes but g2o files are not.

sFaster, 2024-03-29:
Hi.
My english is not good. sorry.

I am testing ba_demo.cpp.
After cheched it work good, moved to my project class.

When I test same code in my class, write different g2o file.

[image]

can you help me?

sFaster, 2024-03-29:
[image]
Left is ba_demo.cpp and right is my class function.

RainerKuemmerle, 2024-03-29:
Somehow your quaternion is not a unit quaternion but all zero.
Your code contains the setIdentity call which should set the quaternion to (0,0,0,1).
I assume, you save the file after optimization? It might be that the translation changes it axis due to this, I did not check this.
Would it be possible to share your project or an minimal example that reproduces the problem?

sFaster, 2024-03-29:
This is before any optimization.
It compiled and ran well, but was not optimized. While looking for the cause, I found that another g2o file was created.

As in the demo, setFix was specified for the first two vertices. Can this be changed during the optimization process?

This is what I confirmed on OSX.

Thank you for your kind reply.

```c++
void System::testG2o() {

	g2o::SparseOptimizer optimizer;
	optimizer.setVerbose(false);
	std::string solverName = "lm_fix6_3";

//	if (DENSE) {
		solverName = "lm_dense6_3";
//	} else {
//#ifdef G2O_HAVE_CHOLMOD
//		solverName = "lm_fix6_3_cholmod";
//#else
//		solverName = "lm_fix6_3";
//#endif
//	}

	g2o::OptimizationAlgorithmProperty solverProperty;
	optimizer.setAlgorithm(
			g2o::OptimizationAlgorithmFactory::instance()->construct(solverName,solverProperty));

	std::vector<Eigen::Vector3d> true_points;
	for (size_t i = 0; i < 500; ++i) {
		true_points.push_back(
				Eigen::Vector3d((g2o::Sampler::uniformRand(0., 1.) - 0.5) * 3,
					g2o::Sampler::uniformRand(0., 1.) - 0.5,
					g2o::Sampler::uniformRand(0., 1.) + 3));
	}

	double focal_length = 1000.;
	Eigen::Vector2d principal_point(320., 240.);

	std::vector<g2o::SE3Quat, Eigen::aligned_allocator<g2o::SE3Quat> > true_poses;
	g2o::CameraParameters* cam_params =
		new g2o::CameraParameters(focal_length, principal_point, 0.);
	cam_params->setId(0);

	if(!optimizer.addParameter(cam_params)) assert(false);

	int vertex_id = 0;
	for (size_t i = 0; i < 15; ++i) {
		Eigen::Vector3d trans(i * 0.04 - 1., 0, 0);
		//printf("trans [ %f %f %f ]\n",trans(0),trans(1),trans(2));
		Eigen::Quaterniond q;
		q.setIdentity();
		g2o::SE3Quat pose(q, trans);
		//std::cout << pose << std::endl;
		g2o::VertexSE3Expmap* v_se3 = new g2o::VertexSE3Expmap();
		v_se3->setId(vertex_id);
		if (i < 2) v_se3->setFixed(true);
		v_se3->setEstimate(pose);
		optimizer.addVertex(v_se3);
		true_poses.push_back(pose);
		vertex_id++;
	}
	int point_id = vertex_id;
	double sum_diff2 = 0;

	std::cout << std::endl;
	std::unordered_map<int, int> pointid_2_trueid;
	std::unordered_set<int> inliers;

	for (size_t i = 0; i < true_points.size(); ++i) {
		g2o::VertexPointXYZ* v_p = new g2o::VertexPointXYZ();
		v_p->setId(point_id);
		v_p->setMarginalized(true);
		v_p->setEstimate(true_points.at(i) +
				Eigen::Vector3d(g2o::Sampler::gaussRand(0., 1),
					g2o::Sampler::gaussRand(0., 1),
					g2o::Sampler::gaussRand(0., 1)));
		int num_obs = 0;
		for (size_t j = 0; j < true_poses.size(); ++j) {
			Eigen::Vector2d z = cam_params->cam_map(true_poses.at(j).map(true_points.at(i)));
			if (z[0] >= 0 && z[1] >= 0 && z[0] < 640 && z[1] < 480) ++num_obs;
		}

		if (num_obs >= 2) {
			optimizer.addVertex(v_p);
			bool inlier = true;
			for (size_t j = 0; j < true_poses.size(); ++j) {
				Eigen::Vector2d z = cam_params->cam_map(true_poses.at(j).map(true_points.at(i)));

				if (z[0] >= 0 && z[1] >= 0 && z[0] < 640 && z[1] < 480) {
					double sam = g2o::Sampler::uniformRand(0., 1.);
					if (sam < 0.005) {
						z = Eigen::Vector2d(Sample::uniform(0, 640), Sample::uniform(0, 480));
						inlier = false;
					}

					double PixelNoise = 3.0;
					z += Eigen::Vector2d(g2o::Sampler::gaussRand(0., PixelNoise),
							g2o::Sampler::gaussRand(0., PixelNoise));

					g2o::EdgeProjectXYZ2UV* e = new g2o::EdgeProjectXYZ2UV();
					e->setVertex(0, dynamic_cast<g2o::OptimizableGraph::Vertex*>(v_p));
					e->setVertex(1, dynamic_cast<g2o::OptimizableGraph::Vertex*>(optimizer.vertices().find(j)->second));
					e->setMeasurement(z);
					e->information() = Eigen::Matrix2d::Identity();
					//e->setRobustKernel(new g2o::RobustKernelHuber);
					e->setParameterId(0, 0);
					optimizer.addEdge(e);
				}
			}

			if (inlier) {
				inliers.insert(point_id);
				Eigen::Vector3d diff = v_p->estimate() - true_points[i];

				sum_diff2 += diff.dot(diff);
			}
			pointid_2_trueid.insert(std::make_pair(point_id, i));
			++point_id;
		}
	}
	std::cout << std::endl;
	optimizer.initializeOptimization();
	optimizer.setVerbose(true);

//	if (STRUCTURE_ONLY) {
//		g2o::StructureOnlySolver<3> structure_only_ba;
//		cout << "Performing structure-only BA:" << endl;
//		g2o::OptimizableGraph::VertexContainer points;
//		for (g2o::OptimizableGraph::VertexIDMap::const_iterator it =
//				optimizer.vertices().begin();
//				it != optimizer.vertices().end(); ++it) {
//			g2o::OptimizableGraph::Vertex* v =
//				static_cast<g2o::OptimizableGraph::Vertex*>(it->second);
//			if (v->dimension() == 3) points.push_back(v);
//		}
//		structure_only_ba.calc(points, 10);
//	}

	optimizer.save("before.g2o");
	std::cout << std::endl;
	std::cout << "Performing full BA:" << std::endl;
	optimizer.optimize(10);
	std::cout << std::endl;
	std::cout << "Point error before optimisation (inliers only): "
		<< sqrt(sum_diff2 / inliers.size()) << std::endl;
	optimizer.save("after.g2o");

	sum_diff2 = 0;
	for(std::unordered_map<int, int>::iterator it = pointid_2_trueid.begin();
			it != pointid_2_trueid.end(); ++it) {
		g2o::HyperGraph::VertexIDMap::iterator v_it = optimizer.vertices().find(it->first);

		if (v_it == optimizer.vertices().end()) {
			std::cerr << "Vertex " << it->first << " not in graph!" << std::endl;
			exit(-1);
		}

		g2o::VertexPointXYZ* v_p = dynamic_cast<g2o::VertexPointXYZ*>(v_it->second);
		if (v_p == 0) {
			std::cerr << "Vertex " << it->first << "is not a PointXYZ!" << std::endl;
			exit(-1);
		}

		Eigen::Vector3d diff = v_p->estimate() - true_points[it->second];
		if (inliers.find(it->first) == inliers.end()) continue;
		sum_diff2 += diff.dot(diff);
	}
	std::cout << "Point error after optimisation (inliers only): "
		<< sqrt(sum_diff2 / inliers.size()) << std::endl;
	std::cout << std::endl;

	optimizer.clear();
}

size_t System::makePoints(FramePtr &curr, std::vector<cv::DMatch> &matches) {
	testG2o();
```

RainerKuemmerle, 2024-03-29:
So after checking the file, the issue got resolved?
setFixed() can be changed before the next round of 
```
optimizer.initializeOptimization();
optimizer.optimize(...);
```

sFaster, 2024-03-29:
I found a way. Thank you.
I don't know the cause, but when I removed the -march=native option from CMakeLists.txt, optimization was performed normally.

The equipment being used is as follows.

osx : Sonoma 14.4.1
compiler:
Apple clang version 15.0.0 (clang-1500.3.9.4)
Target: x86_64-apple-darwin23.4.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

git : 
* (HEAD detached at 20230806_git)

closed 2024-03-29 by sFaster

RainerKuemmerle, 2024-03-29:
Different compiler flags (in particular -march) can effect the memory layout of Eigen types. You need to have the same with respect to those critical flags when compiling g2o and your project.