RainerKuemmerle/g2o #778: Please help me with the rosdep error. I installed libg2o. But still get this error. What should I do?

LEELEE412, 2024-04-01:
$ rosdep install --from-paths src --ignore-src -r -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
scanmatcher: Cannot locate rosdep definition for [libg2o]
graph_based_slam: Cannot locate rosdep definition for [libg2o]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

I searched all the contetns on the internet, but there is no one like me. What is the problem?

RainerKuemmerle, 2024-04-01:
libg2o should be available in ROS Noetic. https://github.com/ros-gbp/libg2o-release
Try to install it directly via apt (assuming you use Ubuntu).

Other than that, I am not maintaining any ROS release in this repository.

LEELEE412, 2024-04-01:
But I'm using ros2 foxy for ubuntu 20.04. I also want to download it via apt. I also want to download it as apt. But I keep trying, and it fails.  Do I have to install Bloom Release?

closed 2025-01-24 by github-actions[bot]