RainerKuemmerle/g2o #781: Prerequisite reading materials and pathway to learning to use the g20 library

Mechazo11, 2024-04-06:
Dear g2o team,

Thank you for your excellent work on the g2o library. I've been using it with ORB-SLAM3 for two years and now aim to update its implementation to be compatible with the latest g2o version. My background in mechanical engineering and initial unfamiliarity with linear algebra and C++ made the learning curve steep, but I've since improved. Could you recommend how a newcomer might start with g2o and any helpful resources or reading materials?

Best,
@Mechazo11

jeremysalwen, 2024-07-30:
Hey @Mechazo11  I don't know about g2o, but if you have success, we definitely would be interested in the patch for the unofficial ORB_SLAM3 community project: https://github.com/jeremysalwen/ORB_SLAM_COMMUNITY

Mechazo11, 2024-07-30:
Hi @jeremysalwen, 

I would be honored to be a part of the community ORB-SLAM3 project.  A while back I created a native ROS 2 Humble package for ORB-SLAM3, [ros2_orb_slam3](https://github.com/Mechazo11/ros2_orb_slam3). I have received some positive feedback regarding it and a major change from the original was I had moved how images were read and passed to the VLSAM node in effort to stream line using multiple datasets. If you would kindly add me as one of its contributors, I will appreciate it greatly.

Regarding g2o, back in April, I did try to upgrade the g2o to its [latest build](https://github.com/RainerKuemmerle/g2o) but between the original ORB-SLAM3 and the latest g2o (April 2023), if I am not mistaken, the library has undergone significant overhaul. Hence I had opened this issue back in April in hopes that if the authors may guide me to some reading materials, I can start to figure out and document how the library worked to figure out the upgrade to the latest build.

With best,
@Mechazo11

RainerKuemmerle, 2024-07-30:
I quickly looked at https://github.com/Mechazo11/ros2_orb_slam3/tree/main/orb_slam3/Thirdparty/g2o
This seems like bundling an excerpt of the g2o codebase there, providing a custom CMake and re-organizing the folder structure.

g2o on the master branch did not have that many changes. But integrating something new into the above project does not seem straight-forward effort due to the aforementioned changes done in the orb-slam project.
Do me, this seems more like a fork with all its pros and cons attached.

For example, in https://github.com/UZ-SLAMLab/ORB_SLAM3/tree/master which is the original upstream, there is one commit which changes some defaults of the Levenberg optimizer class. Maybe there are more changes in the initial commit which would require deeper analysis but yield the nice bundled ORB-SLAM package.

I am not really sure how to support here best. Probably one point would be what you would see as a possible benefit of updating the "forked" g2o?

Mechazo11, 2024-07-30:
Dr. @RainerKuemmerle thank you very much for the insightful analysis.

What I am really looking to achieve is removing the ```modified``` version of g2o that came with the original ORB-SLAM3 with the standard version from your repository since you and a number of developers are actively supporting this library. My end goals are to learn to use this optimization library and also come up with a ROS 2 humble package for this library.

Sir, would you suggest a reading material and or a simple example that I can work with to get started in learning to using this library.