RainerKuemmerle/g2o #819: Segmentation fault (core dumped) of g2o-20230223_git on ubuntu20 or ubuntu22

qhtxj, 2024-07-19:
Hello,
   When I run the samples such as curve_fit, the program run into Segmentation fault. The g2o was already build with "-DBUILD_WITH_MARCH_NATIVE=ON". Or for the other codes build on g2o, they also run into Segmentation fault unless I build them with the older version g2o.

RainerKuemmerle, 2024-07-20:
I would suggest MARCH_NATIVE=OFF for debugging.
Are you running the examples from the build/bin folder?
Can you provide ldd output of your binaries, are the SO used from the corresponding lib folder in build/lib?

Can you run with gdb and provide the stack trace?

qhtxj, 2024-07-23:
Thanks very much for your response. I have found the problem that I installed g2o of two versions and set the "export LD_LIBRARY_PATH" to the older one in the bashrc file. When I use the newer version, I should annotate the setting. I will close the issue.

closed 2024-07-23 by qhtxj