RainerKuemmerle/g2o #839: Release GIL inside optimize?

safijari, 2024-10-31:
Wondering if there any reason to retain the GIL in this function?

https://github.com/RainerKuemmerle/g2o/blob/e8d3ccf3e937d6fb08a21afbe19d8212911b8956/python/core/py_sparse_optimizer.cpp#L47

I'm running into a situation where some times an optimization can take a very long time (in the order of minutes) and because this call does not release the GIL it starves out some background threads meant for telemetry.

RainerKuemmerle, 2024-11-01:
I have to admit that I did so far not think about this that much. Mostly used the Python binding so far to quickly prototype and visualize some data.

Reading the documentation (https://pybind11.readthedocs.io/en/stable/advanced/misc.html#global-interpreter-lock-gil) I believe we can release the GIL.

Would be great if you can test the branch and provide feedback here. Thanks.

safijari, 2024-11-04:
Are wheels saved anywhere that I could grab and install?

RainerKuemmerle, 2024-11-17:
Not for updated g2o. You'd have to build yourself. Probably best done in a python venv.

```
git clone --recurse-submodules https://github.com/miquelmassot/g2o-python.git
cd g2o-python/g2o
git remote set-url origin https://github.com/RainerKuemmerle/g2o.git
git fetch -tp
git co pymem-gil-optimize
cd ..
pip install -v .
```

RainerKuemmerle, 2025-01-01:
I merged the PR now. Let me know in case we see side effects.

closed 2025-01-01 by RainerKuemmerle